linux4sam/meta-atmel #49: please add meta-atmel to the OE metadata index

twoerner, 2015-07-24:
Hi,

Would you be interested in adding your meta-atmel layer to the OpenEmbedded Metadata Index? It make it easier for others to find your work!

http://layers.openembedded.org/layerindex/branch/master/layers/
http://layers.openembedded.org/layerindex/submit/

Best regards,
   Trevor

noglitch, 2015-07-29:
Indeed I'd love it.
In fact I'm heading for it for years but always find the layer not in good shape enough to take the plunge...

noglitch, 2015-07-29:
I've just done it... let's wait for its review by the moderator.

twoerner, 2015-07-30:
Awesome!! I can see that it has shown up in the index:
http://layers.openembedded.org/layerindex/branch/master/layer/meta-atmel/

Thanks :-)

closed 2015-07-30 by twoerner

noglitch, 2015-07-30:
Yes, I'm happy it does! Thanks, bye.