linux4sam/meta-atmel #60: Why does a core-image-minimal need qt?

RobertBerger, 2015-11-10:
layer                 path                                      priority
==========================================================================
meta                  /home/student/poky/meta                   5
meta-yocto            /home/student/poky/meta-yocto             5
meta-yocto-bsp        /home/student/poky/meta-yocto-bsp         5
meta-atmel            /home/student/poky/meta-atmel             10

MACHINE=sama5d3-xplained bitbake core-image-minimal
ERROR: ParseError at /home/student/poky/meta-atmel/recipes-qt/apps/applicationlauncher_1.5.bb:17: Could not inherit file classes/qmake5.bbclass                         

This fixed it, but I think qt recipes should not be required by default:

layer                 path                                      priority
==========================================================================
meta                  /home/student/poky/meta                   5
meta-yocto            /home/student/poky/meta-yocto             5
meta-yocto-bsp        /home/student/poky/meta-yocto-bsp         5
meta-atmel            /home/student/poky/meta-atmel             10
meta-qt5              /home/student/poky/meta-qt5               7

alexandrebelloni, 2015-11-12:
Hi Robert, the following commit should fix it:
https://github.com/linux4sam/meta-atmel/commit/5d474370cbc553a6ab3e24b4d87775c676c51f56

RobertBerger, 2015-11-12:
Hi Alexandre
Did you try this with fido? I forgot to mention that I am playing with fido 1.8.1 at the moment.

git am ~/meta-atmel-patch/patches/0001-Move-QT5-recipes-to-QT5-layer-directory.patch 
Applying: Move QT5 recipes to QT5 layer directory
/home/student/poky/meta-atmel/.git/rebase-apply/patch:194: trailing whitespace.
 
/home/student/poky/meta-atmel/.git/rebase-apply/patch:195: space before tab in indent.
        dd if=/dev/zero of=/dev/fb1
/home/student/poky/meta-atmel/.git/rebase-apply/patch:203: space before tab in indent.
        # Hack to prevent the pipeline from being stuck
/home/student/poky/meta-atmel/.git/rebase-apply/patch:206: space before tab in indent.
        if [ -d /proc/asound/card0 ]
/home/student/poky/meta-atmel/.git/rebase-apply/patch:207: space before tab in indent.
        then
error: recipes-qt/apps/applicationlauncher_1.7.bb: does not exist in index
Patch failed at 0001 Move QT5 recipes to QT5 layer directory
The copy of the patch that failed is found in:
   /home/student/poky/meta-atmel/.git/rebase-apply/patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".

alexandrebelloni, 2015-11-13:
Hi robert, can you try https://github.com/linux4sam/meta-atmel/pull/62 ?

RobertBerger, 2015-11-13:
Hi,

The fido branch in github.com/alexandrebelloni/meta-atmel.git seems to do the trick without qt5. Thanks!
Hopefully this will soon go upstream!

Regards,

Robert

noglitch, 2015-11-13:
I've just merged @alexandrebelloni PR. Thanks for your feedback @RobertBerger .
Bye,

closed 2015-11-13 by noglitch