linux4sam/meta-atmel #65: sama5d2 config files have inconsistencies

ey3ball, 2016-04-18:
The documentation available here :

http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d2XplainedMainPage#Build_Yocto_Poky_rootfs_from_sou

Describes how to rebuild a full linux system for the xplained boardusing the "sama5d2-xplained" config file

However the given example produces images which are hard to use, in particular the resulting u-boot binary ships with an "ubi" command line while the rootfs is provided in .tar.gz format

* It seems odd that the main image would expect an UBI rootfs while there is no flash-based mass storage on the board
* If the bootloader expects an ubi root, an .ubi product file would be better suited than a tarball
* Considering the documentation only describes how to use a system hosted on the sdcard, it should rather reference the sama5d2-sd machine

Regarding the sama5d2-xplained-sd MACHINE,

* It generates binaries + a tarball, it would be useful to add "ext3" to IMAGE_FSTYPES or generate a full .sdcard image.
* By default /etc/inittab is populated with : `S0:12345:respawn:/sbin/getty -L 115200 ttyS0`, however u-boot is setup with baudrate=56700, and boots the kernel with "bootargs=console=ttyS0,57600", therefore once the kernel has booted, the console outputs garbage.

alexandrebelloni, 2016-08-09:
The sama5d2-xplained machine is supposed to have at91bootstrap, u-boot, the kernel and devicetree on the SPI flash then boot on a rootfs on the SD card.

This PR solves that: https://github.com/linux4sam/u-boot-at91/pull/4

Regarding the bitrate. It has been changed consistently to 115200 in the latest versions of at91bootstrap and u-boot. This PR solves that: https://github.com/linux4sam/meta-atmel/pull/70

alexandrebelloni, 2016-08-09:
And now, I've submitted https://github.com/linux4sam/meta-atmel/pull/71 to solve the last issue. It uses wic to generate sdcard image ready to be flashed using dd.

I think we can close that issue once the PR are merged.

noglitch, 2016-08-09:
@alexandrebelloni 's PR taken and merged.
I close this issue.
@ey3ball : thanks for your feedback.
Bye

closed 2016-08-09 by noglitch

Majdoddin, 2016-08-15:
I have a SAMA5D3 xplained board.
Will you please also apply these changes  to this board? sama5d3-xplained-sd.conf and sama5d3-xplained.conf.
Specially, its great if the flash and SD card images are generated automatically.

Moreover, I am interested to configure the system such that the  console operates over the USB port (USB-A, J6), and in my computer I would see it as /dev/ttyACM0. This the behavior for the factory image on the board and also for demo images (see [here](http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d3XplainedMainPage)), but I do not know how to do it with Yocto, or when I build the kernel myself. It would be nice if this is added as an option in conf files. 
thanks