linux4sam/meta-atmel #68: Error by building core minimal image

Majdoddin, 2016-07-29:
Hello, 
I got an error by step 9 of the README file, namely by executing
`bitbake core-image-minimal`
I get 
`...
 fatal error: linux/compiler-gcc6.h: No such file or directory
|  #include gcc_header(__GNUC__)
|                               ^
...`
Apparently there is a problem with compiling the kernel with GCC 6, and  a file `linux/compiler-gcc6.h` is missing. A similar issue is addressed [here](https://github.com/Freescale/meta-fsl-arm/blob/master/recipes-kernel/linux/linux-ls1/Fix-the-compile-issue-under-gcc6.patch) for `meta-fsl-arm`.

I also add the complete output:
```
~/mcg/poky/build-atmel$ bitbake core-image-minimal
WARNING: /home/rooholah/mcg/meta-atmel/recipes-kernel/linux/linux-at91_3.10.bb: Unable to get checksum for linux-at91 SRC_URI entry defconfig: file could not be found                                                                      
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/u-boot/u-boot-at91_2015.01.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/u-boot/u-boot-at91_2014.07.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/u-boot/u-boot-at91_2013.07.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_3.7.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_3.7.2.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_3.7.1.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_git.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/u-boot/u-boot-at91_git.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-qt5/recipes-qt/qt5/qtbase_git.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_3.8.4.bb: base_contains is deprecated, please use bb.utils.contains instead.
NOTE: /home/rooholah/mcg/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_3.8.3.bb: base_contains is deprecated, please use bb.utils.contains instead.
Parsing recipes: 100% |################################################################################| Time: 0:01:28
Parsing of 1964 .bb files complete (0 cached, 1964 parsed). 2544 targets, 395 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.5"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "sama5d3-xplained"
DISTRO            = "poky-atmel"
DISTRO_VERSION    = "2.1+snapshot-20160728"
TUNE_FEATURES     = "arm armv7a vfp thumb            callconvention-hard            cortexa5"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:039f47ad197a9a53109c9f3deadd9c35e62c056d"
meta-atmel        = "master:df10071966a78f0a808216bd02cf961b171cbb0c"
meta-oe           
meta-networking   
meta-python       
meta-ruby         
meta-multimedia   = "master:5c69367d848271cdf3a3bd1a087665f324d3fd34"
meta-qt5          = "master:ecf7db0bd5012162d5fff5cb25f9a07a1a6e084a"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/1.1/x86_64-nativesdk-libc.tar.bz2;sha256sum=84e89ed675156d13a27fe56e8433f5fa9662802349bb0bbe9e6eba577fc140d3
--2016-07-28 15:38:53--  http://downloads.yoctoproject.org/releases/uninative/1.1/x86_64-nativesdk-libc.tar.bz2
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)... 198.145.20.127
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|198.145.20.127|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2481482 (2.4M) [application/octet-stream]
Saving to: ‘/home/rooholah/mcg/poky/build-atmel/downloads/uninative/84e89ed675156d13a27fe56e8433f5fa9662802349bb0bbe9e6eba577fc140d3/x86_64-nativesdk-libc.tar.bz2’

2016-07-28 15:38:57 (585 KB/s) - ‘/home/rooholah/mcg/poky/build-atmel/downloads/uninative/84e89ed675156d13a27fe56e8433f5fa9662802349bb0bbe9e6eba577fc140d3/x86_64-nativesdk-libc.tar.bz2’ saved [2481482/2481482]

Initialising tasks: 100% |#############################################################################| Time: 0:00:03
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-at91-4.1+gitAUTOINC+00ea8f3f67-r0 do_compile: oe_runmake failed
ERROR: linux-at91-4.1+gitAUTOINC+00ea8f3f67-r0 do_compile: Function failed: do_compile (log file is located at /home/rooholah/mcg/poky/build-atmel/tmp/work/sama5d3_xplained-poky-linux-gnueabi/linux-at91/4.1+gitAUTOINC+00ea8f3f67-r0/temp/log.do_compile.13779)
ERROR: Logfile of failure stored in: /home/rooholah/mcg/poky/build-atmel/tmp/work/sama5d3_xplained-poky-linux-gnueabi/linux-at91/4.1+gitAUTOINC+00ea8f3f67-r0/temp/log.do_compile.13779
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 zImage CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd LD=arm-poky-linux-gnueabi-ld.bfd
[... 73 lines cut ...]
|   HOSTCC  scripts/dtc/util.o
|   SHIPPED scripts/dtc/dtc-lexer.lex.c
|   SHIPPED scripts/dtc/dtc-parser.tab.h
|   SHIPPED scripts/dtc/dtc-parser.tab.c
|   HOSTCC  scripts/dtc/dtc-lexer.lex.o
|   HOSTCC  scripts/dtc/dtc-parser.tab.o
|   HOSTLD  scripts/dtc/dtc
| /home/rooholah/mcg/poky/build-atmel/tmp/work-shared/sama5d3-xplained/kernel-source/Makefile:555: recipe for target 'scripts' failed
| make[2]: *** [scripts] Error 2
| make[2]: *** Waiting for unfinished jobs....
|   UPD     include/config/kernel.release
| Makefile:146: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/rooholah/mcg/poky/build-atmel/tmp/work/sama5d3_xplained-poky-linux-gnueabi/linux-at91/4.1+gitAUTOINC+00ea8f3f67-r0/temp/log.do_compile.13779)
ERROR: Task /home/rooholah/mcg/meta-atmel/recipes-kernel/linux/linux-at91_4.1.bb:do_compile (/home/rooholah/mcg/meta-atmel/recipes-kernel/linux/linux-at91_4.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1229 tasks of which 8 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/rooholah/mcg/meta-atmel/recipes-kernel/linux/linux-at91_4.1.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

alexandrebelloni, 2016-08-08:
That one should be solved in the kernel repository instead of keeping a patch in meta-atmel

alexandrebelloni, 2016-08-09:
IT should be solved by: https://github.com/linux4sam/linux-at91/commit/aa504b4448f342ff952aaea9eac7e80ae19154d8

noglitch, 2016-08-09:
Solved as @alexandrebelloni pointed out.
Thanks @Majdoddin for your feedback.

closed 2016-08-09 by noglitch