linux4sam/meta-atmel #74: automatic generation of SD card and flash image - console over ttyACM0

Majdoddin, 2016-08-15:
Hello, 
I am writing mostly with regard to issue [#65](https://github.com/linux4sam/meta-atmel/issues/65), which is closed now.
Will you please apply the changes to the board SAMA5D3 Xplained? (sama5d3-xplained.conf and sama5d3-xplained-sd.conf)
Specially it would be very nice if the flash and SD card images are generated automatically.

Moreover, I want to configure my system for this board, such that the console operates over the USB port (named USB-A, J6 on the board) and I see it on my computer as /dev/ttyACM0. That is how the factory image on the board does, and also the demo images do it that way ([see here](http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d3XplainedMainPage#USB_device_port)) . But I do not know how to do it using Yocto, or when I build the kernel myself. It would be great if that (with other console settings like baudrate) is added as an option in the config files.
Thanks

alexandrebelloni, 2016-08-16:
https://github.com/linux4sam/meta-atmel/pull/75 should properly generate sd card images for sama5d3 xplained

Regarding getting the console on ttyGS0 (that is the name on the xplained side), it is already an option that is present in `sama5d3-xplained-sd.conf`:

`SERIAL_CONSOLES ?= "115200;ttyS0 115200;ttyGS0"`

There is nothing more we can do.

The relevant kernel configuration is:

```
CONFIG_USB_GADGET=y
CONFIG_USB_ATMEL_USBA=m
CONFIG_USB_CONFIGFS=m
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
CONFIG_USB_MASS_STORAGE=m
CONFIG_USB_G_SERIAL=m
```

Majdoddin, 2016-08-17:
Thank you very much for #75 (Is it also possible to generate image for the flash?).

Regarding the console, something is still missing:
Using the [Demo image](http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d3XplainedMainPage#Demo_archives), when USB cable is plugged (port USB-A, J6),  I got /dev/ttyACM0 on my computer (/dev/ttyGS0 on board side) which I can use  with my terminal software to connect to the board console. **And** if also a USB to TTL serial cable is pluged (port J23), I also got /dev/ttyUSB0 on my computer (/dev/ttyS0 on board side) which I can use  the same way.
**But**  if I build with Yocto and meta-atmel, I get nothing with the USB cable. Only with the USB to TTL cable I have access to console. 
 `dmesg` of my computer with Demo image on board:

...
[736608.804874] usb 5-2: new high-speed USB device number 34 using ehci-pci
[736608.937474] usb 5-2: New USB device found, idVendor=03eb, idProduct=6124
[736608.937482] usb 5-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[736608.937931] cdc_acm 5-2:1.0: This device cannot do calls on its own. It is not a modem. 

[736608.938052] cdc_acm 5-2:1.0: ttyACM0: USB ACM device
...
[777949.178538] usbcore: registered new interface driver ftdi_sio
[777949.178594] usbserial: USB Serial support registered for FTDI USB Serial Device
[777949.179009] ftdi_sio 7-1:1.0: FTDI USB Serial Device converter detected
[777949.179140] usb 7-1: Detected FT232RL
[777949.179145] usb 7-1: Number of endpoints 2
[777949.179150] usb 7-1: Endpoint 1 MaxPacketSize 64
[777949.179155] usb 7-1: Endpoint 2 MaxPacketSize 64
[777949.179159] usb 7-1: Setting MaxPacketSize 64
[777949.182169] usb 7-1: FTDI USB Serial Device converter now attached to ttyUSB0
[778677.926385] usb 7-1: USB disconnect, device number 2
[778677.926666] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
...

 I also add the `dmesg` of linux on the board in both cases for comparision.
The `dmesg`of Demo linux on board (the last part starting with tag [ 3.960000] , is `udev` that set ups `g_serial gadget` and seems important):

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.0-linux4sam_5.3 (ldesroches@ibiza) (gcc version 5.2.0 (GCC) ) #1 Sat Apr 16 11:28:00 CEST 2016
[    0.000000] CPU: ARMv7 Processor [410fc051] revision 1 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: SAMA5D3 Xplained
[    0.000000] cma: Reserved 64 MiB at 0x28000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c06e20c4, node_mem_map cfdda000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 186828K/262144K available (4985K kernel code, 179K rwdata, 1664K rodata, 188K init, 159K bss, 9780K reserved, 65536K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
                   lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc0686928   (6651 kB)
                     .init : 0xc0687000 - 0xc06b6000   ( 188 kB)
                     .data : 0xc06b6000 - 0xc06e2ed8   ( 180 kB)
                      .bss : 0xc06e2ed8 - 0xc070ad3c   ( 160 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] clocksource pit: mask: 0xfffffff max_cycles: 0xfffffff, max_idle_ns: 14479245754 ns
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
[    0.000000] Console: colour dummy device 80x30
[    0.050000] Calibrating delay loop... 351.43 BogoMIPS (lpj=1757184)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 353 lines cut ...]
usbhid: USB HID core driver
iio iio:device0: Resolution used: 12 bits
iio iio:device0: ADC Touch screen is disabled.
NET: Registered protocol family 17
can: controller area network core (rev 20120528 abi 9)
NET: Registered protocol family 29
can: raw protocol (rev 20120528)
can: broadcast manager protocol (rev 20120528 t)
can: netlink gateway (rev 20130117) max_hops=1
input: gpio_keys as /devices/soc0/gpio_keys/input/input0
at91_rtc fffffeb0.rtc: setting system clock to 2016-08-16 08:39:09 UTC (1471336749)
ALSA device list:
  No soundcards found.
EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p2): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs (mmcblk0p2): recovery complete
EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
VFS: Mounted root (ext4 filesystem) on device 179:2.
devtmpfs: mounted
Freeing unused kernel memory: 192K (c0688000 - c06b8000)
udevd[651]: starting version 3.2
random: udevd urandom read with 78 bits of entropy available
udevd[652]: starting eudev-3.2
EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
random: nonblocking pool is initialized

alexandrebelloni, 2016-08-20:
Building the image for the flash is done using the sama5d3-xplained machine configuration.

It is difficult to help you further regarding your issue with ttyGS0 because I don't know what you include in your image.
I've tested atmel-xplained-demo-image on krogoth and this works fine.

Majdoddin, 2016-08-23:
I know about sama5d3-xplained machine configuration, but it does not generate a single flash image, a wic file. By contrast, sama5d**2**-xplained does it. 

And abut the ttyGS0 issue, now I understand it, many thanks.

alexandrebelloni, 2016-08-23:
It generates a .ubi file that is directly flashable on the NAND flash through sam-ba (see http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d3XplainedMainPage#Program_rootfs_UBI_file)

sama5d2 doesn't have NAND flash and only has emmc and SD card. The generated wic image is supposed to be written on an SDcard instead of untaring on the SDcard (as in http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d2XplainedMainPage#Flash_the_demo).

Majdoddin, 2016-08-23:
Now I see, Thank you very much.

closed 2016-08-25 by noglitch