linux4sam/meta-atmel #76: sama5d4ek use soft float toolchain

Blackrose, 2016-08-19:
Hi, i use yocto to build filesystem for sama5d4ek board.And follow your instruction to setup env values, and it build success.Though i found the all packages and sdk is armel(soft float).Here is my build configuration

```
Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "sama5d4ek"
DISTRO            = "poky-atmel"
DISTRO_VERSION    = "2.0.2"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard cortexa5"
TARGET_FPU        = "vfp-neon"
meta              
meta-yocto        
meta-yocto-bsp    = "jethro:40376446904ae3529be41737fed9a0b650ed167d"
meta-oe           
meta-networking   
meta-python       
meta-ruby         
meta-multimedia   = "jethro:8ab04afbffb4bc5184cfe0655049de6f44269990"
meta-atmel        = "jethro:4765d7064e4916784c15095347eda21cc10aabb4"
meta-qt5          = "jethro:ea37a0bc987aa9484937ad68f762b4657c198617"
```

Here is my question, I found the variable DEFAULTTUNE is cortexa5thf-neon in file sama5d4.inc. But build packages is armel, not armhf.Is something  i forgot config? Thanks!

alexandrebelloni, 2016-08-20:
callconvention-hard in TUNE_FEATURES indicates that you properly build for armhf. On my side, packages are properly generated for armhf. What makes you think they are armel?

Blackrose, 2016-08-21:
Thanks your reply. I had checked it again.Those binary files are armhf in filesystem tarball file.On building same time, i choose to build package as deb format.And the deb packages are use "armel" as suffix of package name, not the "armhf".Is it right? And why not use armhf?

alexandrebelloni, 2016-08-21:
Well, this has changed in krogoth (see http://git.openembedded.org/openembedded-core/commit/meta/classes/package_deb.bbclass?id=8d042ea4e755cb0bb28b88333e10e04ec4e86a36) and that actually was an issue (see http://git.openembedded.org/openembedded-core/commit/meta/recipes-devtools/dpkg?id=b01a01ff47e09da4aaa2db992380ca0498f0e5ae).

I'd say as long as your packages can be installed and are linked to the proper libc (armhf), the naming doesn't matter. You also have the possibility to update to krogoth that should be working fine

Blackrose, 2016-08-23:
Thanks again!I will update my oe-core code.

closed 2016-08-25 by noglitch