linux4sam/meta-atmel #81: Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

smxcyp, 2016-11-25:
atmel@ubuntu:~/my_dir/poky/build-atmel$ bitbake atmel-qt5-demo-image
Parsing recipes: 100% |#########################################| Time: 00:00:37
Parsing of 1934 .bb files complete (0 cached, 1934 parsed). 2490 targets, 362 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "sama5d2-xplained"
DISTRO            = "poky-atmel"
DISTRO_VERSION    = "2.1.2"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard       cortexa5"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:3bf928a3b6354bc09c87fcbf9e3972c8d368aaa3"
meta-atmel        = "krogoth:af0fb3550bba8bbcdecef9e1d40a435a83b74335"
meta-oe           
meta-networking   
meta-python       
meta-ruby         
meta-multimedia   = "krogoth:851a064b53dca3b14dd33eaaaca9573b1a36bf0e"
meta-qt5          = "krogoth:2b1871f0d139dc3caaa779a32a1931409c245a36"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: qtbase-5.6.2+gitAUTOINC+b4ada3f0d8-r0 do_patch: Command Error: 'quilt --quiltrc /home/atmel/my_dir/poky/build-atmel/tmp/sysroots/x86_64-linux/etc/quiltrc push' exited with 1  Output:
Applying patch 0001-Fix-building-without-OpenGL-support.patch
patching file src/widgets/kernel/qwidgetbackingstore.cpp
Hunk #1 FAILED at 61.
1 out of 1 hunk FAILED -- rejects in file src/widgets/kernel/qwidgetbackingstore.cpp
Patch 0001-Fix-building-without-OpenGL-support.patch can be reverse-applied
ERROR: qtbase-5.6.2+gitAUTOINC+b4ada3f0d8-r0 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /home/atmel/my_dir/poky/build-atmel/tmp/work/cortexa5hf-neon-poky-linux-gnueabi/qtbase/5.6.2+gitAUTOINC+b4ada3f0d8-r0/temp/log.do_patch.24614
ERROR: Task 646 (/home/atmel/my_dir/meta-qt5/recipes-qt/qt5/qtbase_git.bb, do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 782 tasks of which 765 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/atmel/my_dir/meta-qt5/recipes-qt/qt5/qtbase_git.bb, do_patch
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

alexandrebelloni, 2016-11-25:
should be solved by https://github.com/linux4sam/meta-atmel/pull/82

noglitch, 2016-11-25:
#82 now merged in meta-atmel krogoth branch.
Thanks @smxcyp  for the feedback and @alexandrebelloni for the quick fix.
Best regards,

closed 2016-11-25 by noglitch