linux4sam/meta-atmel #91: nano.specs support

hot-dog2017, 2017-09-11:
Hi,

Im trying to re-compile binaries for SAM-BA using the Cross compiler exposed by Yocto.

Ive found that there is no support for nano.specs. Is this support to be added for Atmel targets?

Ive downloaded example recipes for "newlib" ie https://github.com/jurobystricky/meta-zephyr/tree/master/recipes-core/newlib

Id like to use the CC generated by Yocto rather that the downloaded version. Could someone help please.

Thanking you in advance..

alexandrebelloni, 2017-09-14:
OE/Yocto doesn't have any support to generate a newlib based toolchain and use it to generate a linux image.

What you are pointing to is a fork of meta-zephyr which is actually a layer used to buld an SDK for zephyr (not Linux).

closed 2017-09-15 by noglitch

hot-dog2017, 2017-09-19:
Your correct, the meta-zephr newlib support ive already tried to incorporate into my SDK but failed to.

There is very little on adding support for newlib. Is there any info / examples someone help with?..

alexandrebelloni, 2017-09-19:
oe-core only has support for glibc and musl. I don't think anybody has added support for newlib.

hot-dog2017, 2017-09-20:
..does body know how i would add newlib support?