linux4sam/meta-atmel #95: Illegal charaster in kernel module with ipk

ramok, 2018-01-15:
I'm trying to build with linux-at91_4.4.bb kernel, and get error:

> ERROR: linux-at91-4.4+gitAUTOINC+061f02b7aa-r0.0 do_package_write_ipk: Error executing a python function in exec_python_func() autogenerated:
>
> The stack trace of python calls that resulted in this exception/failure was:
> File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
>      0001:
>  *** 0002:do_package_ipk(d)
>      0003:
> File: '/work/meta/classes/package_ipk.bbclass', lineno: 87, function: do_package_ipk
>      0083:
>      0084:    os.chdir(oldcwd)
>      0085:
>      0086:    if error:
>  *** 0087:        raise error
>      0088:}
>      0089:do_package_ipk[vardeps] += "ipk_write_pkg"
>      0090:do_package_ipk[vardepsexclude] = "BB_NUMBER_THREADS"
>      0091:
> Exception: subprocess.CalledProcessError: Command 'PATH="/work/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/work/scripts:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot/usr/bin/crossscripts:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot-native/usr/sbin:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot-native/usr/bin:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot-native/sbin:/work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/recipe-sysroot-native/bin:/work/bitbake/bin:/work/build/tmp/hosttools" opkg-build kernel-module-snd-soc-wm8904-4.4.26-linux4sam_5.5 /work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/deploy-ipks/sama5d2_roadrunner' returned non-zero exit status 1
>
> Subprocess output:
> kernel-module-snd-soc-wm8904-4.4.26-linux4sam_5.5
> *** Error: Package name  contains illegal characters, (other than [a-z0-9.+-])
>
> opkg-build: Please fix the above errors and try again.
>
> ERROR: linux-at91-4.4+gitAUTOINC+061f02b7aa-r0.0 do_package_write_ipk: Function failed: do_package_ipk
> ERROR: Logfile of failure stored in: /work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi/linux-at91/4.4+gitAUTOINC+061f02b7aa-r0.0/temp/log.do_package_write_ipk.11726
> ERROR: Task (/work/meta-atmel/recipes-kernel/linux/linux-at91_4.4.bb:do_package_write_ipk) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 1356 tasks of which 1105 didn't need to be rerun and 1 failed.
> 
> Summary: 1 task failed:
>   /work/meta-atmel/recipes-kernel/linux/linux-at91_4.4.bb:do_package_write_ipk
> Summary: There were 18 WARNING messages shown.
> Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

Looks like ipkg did't like 'inux4sam_5.5' in kernel module name.
How I understand problem in linux-at91_4.4.bb with:
> SRCREV = "linux4sam_5.5"
I search how to fix ot or for workaround..

alexandrebelloni, 2018-01-15:
Hi,

I think you can have a look at https://github.com/linux4sam/meta-atmel/commit/555b475b15c17f2e60d61e6f2440bd48ff14e7a8 which fixes something similar.

ramok, 2018-01-15:
thanks, this helps :)

closed 2018-01-15 by ramok