linux4sam/meta-atmel #96: ./scripts/get_sram_size.sh: bc: not found

ramok, 2018-01-16:
I have this error:
> |   LD        sama5d2_roadrunner-dataflashboot-linux--dt-3.8.9.elf
> | ./scripts/get_sram_size.sh: 1: ./scripts/get_sram_size.sh: bc: not found
> | Makefile:338: recipe for target 'ChkFileSize' failed
> | make: *** [ChkFileSize] Error 3 
> | ERROR: oe_runmake failed 
> | WARNING: exit code 1 from a shell command. 
> | ERROR: Function failed: do_compile (log file is located at /work/build/tmp/work/sama5d2_roadrunner-poky-linux-gnueabi
> /at91bootstrap/3.8+gitAUTOINC+d7a52fbf2b-r0.0/temp/log.do_compile.20910)
> ERROR: Task (/work/meta-atmel/recipes-bsp/at91bootstrap/at91bootstrap_git.bb:do_compile) failed with exit code '1'

Adding
> DEPENDS += "bc-native"

to at91bootstrap*.bb helps

alexandrebelloni, 2018-01-16:
Hi,

I believe this is fixed in https://github.com/linux4sam/meta-atmel/commit/20a619cdff5e00e4c3d3c34402730d5fe44867d4

MyleneJ, 2018-01-16:
Hi,

Thank you for reporting this issue.

This error should be already fixed with the following commit:
https://github.com/linux4sam/meta-atmel/commit/20a619cdff5e00e4c3d3c34402730d5fe44867d4

Which branch did you try? Rocko? Last commit?
Could you check if this commit is available under your meta-atmel layer?

Thank you in advance,
Best regards,
Mylène Josserand

ramok, 2018-01-16:
My bad. It's was linux4sam_5.5. I switch to "rocko" and see this patches. Thanks!

closed 2018-01-16 by ramok