linux4sam/meta-atmel #102: arm-poky-linux-gnueabi-g++: error: cec-follower.cpp: No such file or directory

AleFachini, 2018-04-19:
Hello:
So i was following the readme file from this meta layer for yocto-poky and all went good up to step:

11/ Build Atmel demo images
bitbake atmel-qt5-demo-image

I added the lines in conf/local.conf

then started build, and got this error. Ill put the whole log now:

> 
> DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
> DEBUG: Executing shell function do_compile
> NOTE: make -j 1
> make  all-recursive
> make[1]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build'
> Making all in v4l-utils-po
> make[2]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/v4l-utils-po'
> make[2]: Nothing to be done for 'all'.
> make[2]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/v4l-utils-po'
> Making all in libdvbv5-po
> make[2]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/libdvbv5-po'
> make[2]: Nothing to be done for 'all'.
> make[2]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/libdvbv5-po'
> Making all in lib
> make[2]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib'
> Making all in libv4lconvert
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4lconvert'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4lconvert'
> Making all in libv4l2
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l2'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l2'
> Making all in libv4l1
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l1'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l1'
> Making all in libv4l2rds
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l2rds'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l2rds'
> Making all in libv4l-mplane
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l-mplane'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libv4l-mplane'
> Making all in libdvbv5
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libdvbv5'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib/libdvbv5'
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib'
> make[3]: Nothing to be done for 'all-am'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib'
> make[2]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/lib'
> Making all in utils
> make[2]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils'
> Making all in libv4l2util
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/libv4l2util'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/libv4l2util'
> Making all in libmedia_dev
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/libmedia_dev'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/libmedia_dev'
> Making all in decode_tm6000
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/decode_tm6000'
> make[3]: Nothing to be done for 'all'.
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/decode_tm6000'
> Making all in ivtv-ctl
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/ivtv-ctl'
[... 52 lines cut ...]
> make[3]: Entering directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/cec-follower'
> ../../../v4l-utils-1.12.3/utils/cec-ctl/msg2ctl.pl 2 ../../../v4l-utils-1.12.3/include/linux/cec.h ../../../v4l-utils-1.12.3/include/linux/cec-funcs.h >cec-table.h
> arm-poky-linux-gnueabi-g++  -march=armv5e -marm --sysroot=/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../../v4l-utils-1.12.3/utils/cec-follower -I../..   -I../../../v4l-utils-1.12.3/lib/include -Wall -Wpointer-arith -D_GNU_SOURCE -I../../../v4l-utils-1.12.3/include   -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0=/usr/src/debug/v4l-utils/1.12.3-r0 -fdebug-prefix-map=/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/recipe-sysroot=  -fvisibility-inlines-hidden -c -o cec-follower.o cec-follower.cpp
> **arm-poky-linux-gnueabi-g++: error: cec-follower.cpp: No such file or directory**
> **arm-poky-linux-gnueabi-g++: fatal error: no input files**
> compilation terminated.
> ERROR: oe_runmake failed
> Makefile:556: recipe for target 'cec-follower.o' failed
> make[3]: *** [cec-follower.o] Error 1
> make[3]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils/cec-follower'
> Makefile:468: recipe for target 'all-recursive' failed
> make[2]: *** [all-recursive] Error 1
> make[2]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build/utils'
> Makefile:574: **recipe for target 'all-recursive' failed**
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory '/home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/build'
> Makefile:501: recipe for target 'all' failed
> make: *** [all] Error 2
> WARNING: exit code 1 from a shell command.
> **ERROR: Function failed: do_compile** (log file is located at /home/afachini/poky/build-atmel/tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/temp/log.do_compile.26453)
> 

Can you tell me what might be wrong?, the Hardware is SAM9x35 Ronetix Evaluation Kit with LCD. So the MACHINE is "at91sam9x5ek"

Regards.-

alexandrebelloni, 2018-04-25:
This seems like an issue with your filesystem. I've tested an I correctly have
tmp/work/armv5e-poky-linux-gnueabi/v4l-utils/1.12.3-r0/v4l-utils-1.12.3/utils/cec-follower/cec-follower.cpp

So if the tarball passes the checksum, I'm not sure why this file would not be extracted properly.

AleFachini, 2018-04-25:
Well, I also have that file at that directory, so maybe is somenthing not pointing to it. I cant find it yet...

alexandrebelloni, 2018-04-25:
Ok, something is wrong there, my do_compile log ends up with:

-o cec-follower.o ../../../v4l-utils-1.12.3/utils/cec-follower/cec-follower.cpp

while yours have

-o cec-follower.o cec-follower.cpp

Can you try `bitbake -c cleansstate v4l-utils; bitbake v4l-utils`

TolentinoDev, 2018-06-04:
I'm getting a similar error @AleFachini  were you able to solve this problem?

moham96, 2018-07-27:
change this line in v4l-utils recipe
`inherit autotools gettext pkgconfig`
to 
`inherit autotools-brokensep gettext pkgconfig`

noglitch, 2018-08-20:
@alexandrebelloni I still see the first line indicated above in current:
meta-oe/recipes-multimedia/v4l2apps/v4l-utils_1.12.3.bb of sumo meta-openembedded. Does it mean that this change is not needed?

alexandrebelloni, 2018-10-04:
I'm still not sure brokensep is needed there but this can mostly safely be added in a bbappend if the issue persists in sumo.

closed 2019-06-26 by RazvanStef

ganhuaren, 2019-11-28:
thank you. fix the same error

johnwang9, 2020-01-01:
whose suggestion you applied to solve the problem?  I look forward to your reply , thanks!