linux4sam/meta-atmel #111: Yocto build for machine sama5d27-som1-ek-sd fails to build at do_image_wic

martintownsend, 2018-10-23:
I've tried building core-image-minimal using the sumo branch of meta-atmel but it fails at do_image_wic

| output: install: cannot stat '/home/rufilla/projects/luminite/luminite-build-sys/data/build-areas/build-area-sumo/build/tmp/deploy/images/sama5d27-som1-ek-sd/sama5d27_som1_ek.itb': No such file or directory

I've checked and no itb file exists in the deploy directory.

After looking through meta-atmel I think this error occurs because of
IMAGE_BOOT_FILES = "BOOT.BIN u-boot.bin sama5d27_som1_ek.itb"
in the machine configuration file.

If I revert the latest commit, 45cf7c87756cba8449d47fd348b189e76a5915b9, so that the machine configuration now contains:
IMAGE_BOOT_FILES = "BOOT.BIN u-boot.bin zImage \
	zImage-at91-sama5d27_som1_ek.dtb;at91-sama5d27_som1_ek.dtb \
	"
It builds fine.

So should it be creating the itb file (which I assume means the kernel and device tree are in a FIT image) or was the commit incorrect for this machine and it should be using zImage? 

I would prefer to use FIT images if possible so what would I need to do to ensure this machine builds FIT images?

texierp, 2018-10-23:
Hi @martintownsend,

Indeed, it seems you have to build **dt-overlay-at91** package, see [1] for the demo image.

Pierre-Jean

[1] https://github.com/linux4sam/meta-atmel/commit/12b7df3b7f6d3dfd6e674b7890097ec45e90db77#diff-1206dd7427dd942bbb0746408957c18c

martintownsend, 2018-10-23:
Thanks Pierre-Jean.

This worked, I don't need DT overlays so I've created a new machine configuration and set IMAGE_BOOT_FILES back to zImage and dtb files.  

I'll see if I can create a FIT image using kernel-fitimage.bbclass.

closed 2018-10-23 by martintownsend