linux4sam/meta-atmel #124: Yocto build for sama5d27-som1-ek-sd fails to boot into kernel

ainthalux, 2019-03-01:
Hello,

I am currently working with the Atmel/Microchip SAMA5D27-SOM1-EK demo board and was able to get the provided Linux4SAM 6.0 demo image to successfully boot from SD card.  I then followed the readme file from meta-atmel layer for step by step Yocto build procedure and the build completed successfully.  I then flashed the SD card via SAM-BA with the generated .wic file and it completed successfully as well.  However, whenever I attempt to boot up into the built image, it fails to load the kernel.

SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SD: Card Capacity: High or Extended
SD: Specification Version 3.0X
SD/MMC: Done to load image
<debug_uart> 

U-Boot 2018.07-linux4sam_6.0 (Jan 23 2019 - 14:27:07 +0000)

CPU: SAMA5D27 1G bits DDR2 SDRAM
Crystal frequency:       24 MHz
CPU clock        :      492 MHz
Master clock     :      164 MHz
DRAM:  128 MiB
MMC:   sdio-host@a0000000: 0, sdio-host@b0000000: 1
Loading Environment from FAT... OK
In:    serial@f8020000
Out:   serial@f8020000
Err:   serial@f8020000
Net:   eth0: ethernet@f8008000
Hit any key to stop autoboot:  0 
** Unable to read file at91-sama5d27_som1_ek.dtb **
** Unable to read file zImage **
=> 

I see that the machine configuration file for sama5d27-som1-ek-sd contains:

KERNEL_DEVICETREE = " \
              at91-sama5d27_som1_ek.dtb \
              "

IMAGE_BOOT_FILES = "BOOT.BIN u-boot.bin sama5d27_som1_ek.itb"

MACHINE_ESSENTIAL_EXTRA_RDEPENDS = " dt-overlay-at91"

Why would it try to read file at91-sama5d27_som1_ek.dtb and not sama5d27_som1_ek.itb?  Perhaps I'm missing something.

RazvanStef, 2019-03-01:
Hello,

Unfortunately the Yocto generated image does not contain the correct u-boot environment. This will be corrected in the next release. 

You can copy the uboot.env file for the release image on the boot partition of the SD card. 

Best regards,
Razvan

ainthalux, 2019-03-01:
Hello Razvan,

Thank you for your response.  Yes, that is unfortunate but it provided some clarity for me.  I copied the uboot.env file from the released image onto my SD card and it successfully booted into the demo image.  I'll keep a lookout for the next release.

Thanks!

RazvanStef, 2019-03-11:
I've pushed to master 3 patches that will add the u-boot environment from the release to the images. You can cherry-pick them on sumo branch:
ca40be3 u-boot-at91: Generate release u-boot environment
6f126bc u-boot-at91: Add environment to u-boot-at91_2018.07
ce43196 machine: Add uboot.env to wic  #image

Best regards,
Razvan

closed 2019-03-11 by RazvanStef

ainthalux, 2019-04-04:
Hello Razvan,

I just wanted to let you know that I have applied the 3 patches to the image build and was able to get the demo image to boot up successfully.

Thanks!

RazvanStef, 2019-04-04:
Thanks for the feedback.

Regards,
Razvan