linux4sam/meta-atmel #131: qtbase patch error on Thud branch

CleberPeter, 2019-05-16:
I get the following erros when apply the patch's from qtbase on Thud branch:

`Applying patch 0004-Provide-access-to-linuxfb-dri-fd-through-platform.patch
patching file src/platformsupport/fbconvenience/qfbscreen_p.h
patching file src/plugins/platforms/linuxfb/qlinuxfbdrmscreen.cpp
patching file src/plugins/platforms/linuxfb/qlinuxfbdrmscreen.h
patching file src/plugins/platforms/linuxfb/qlinuxfbintegration.cpp
Hunk #1 FAILED at 172.
1 out of 1 hunk FAILED -- rejects in file src/plugins/platforms/linuxfb/qlinuxfbintegration.cpp
patching file src/plugins/platforms/linuxfb/qlinuxfbscreen.h
Patch 0004-Provide-access-to-linuxfb-dri-fd-through-platform.patch does not apply (enforce with -f)`

*Using patch's from master branch the error is solved.

meta-qt5: On Thud branch the version of QT is 5.11.

texierp, 2019-05-16:
Hi @CleberPeter,

Can you check this PR https://github.com/linux4sam/meta-atmel/pull/132

And let me know if it is sufficient for you.

/Pierre-Jean

CleberPeter, 2019-05-16:
Yes, this solve. 

Thanks.

PR #132 (2019-05-16): [thud] : cherry-pick (qtbase: Get it building with QT 5.12)
Fixes #131 

(cherry picked from commit fc5d6882a493a937b15f96a9daadf7057490d02a)
closed 2019-05-16 by RazvanStef