linux4sam/meta-atmel #136: EGL Implementation

CleberPeter, 2019-05-24:
I've been trying to compile Qt Webengine for sama5d27 through the Thud branch without success a few days :(

I checked through the Qt page that one of the requirements of it is OpenGL ES 2.0. I also noticed that the recipe qtbase_%.bbappend in this layer removes the EGLFS plugin if x11 is not defined and unconditionally remove the virtual/egl, both of which are responsible for the interface between OpenGL and the adopted window system.

With a recipe of its own I removed the above definitions and added "gles2" and "eglfs" to PACKAGECONFIG, but I still get errors regarding the atmel-hlcdc driver. Looking again at the Qt page there is a comment regarding the need for each vendor to develop their own adaptation to support the EGL.

Anyway, is there support for EGL for the SAMA5 family? Is it possible to use Qt Webengine with the same?

Could someone suggest me some material (book, presentation, source code, anything) to read about the graphical issues related to linux embedded, I read several materials and so far I have not been able to understand several aspects on that topic.

Thank you.

alexandrebelloni, 2019-05-24:
There is no GPU on the sama5 family, this means that without a software implementation of OpenGL (like mesa), you will not get any EGL implementation.

CleberPeter, 2019-05-25:
Thanks for your answer.

The PREFERRED_PROVIDER_virtual/egl as the default-providers.inc file is assigned as "mesa", so when adding virtual/egl to PACKAGECONFIG I should get success in my build right ?

Besides the "gles2" that enables opengl, what other packages should I add to the qtbase PACKAGECONFIG?

codrin989, 2020-06-11:
Hi @CleberPeter . Have you been able to compile the QT WebEngine?

joshua-henderson, 2020-06-11:
@CleberPeter I can't directly answer your immediate build question at the moment. But (and don't take this to mean if you do get a build going it won't be fine) do know that if you do indeed use opengl anywhere in your path, as noted, you will be using the software implementation of opengl (mesa) and this is basically a no-go by all measures.

I tell you what, I am going to close this issue now.  Please re-open if this issue persists or if there is a misunderstanding.

closed 2020-06-11 by joshua-henderson