linux4sam/meta-atmel #139: Cannot see the 4.5" display connected to the SAMA5D27 SOM1-ek board. I am using the linux4sam-poky-sama5d27_som1_ek_video-6.0.img.bz2 image on the SD card. Do I need to rebuild the image? Thank you.

TMScottSafety, 2019-06-25:


RazvanStef, 2019-06-25:
Hello,

What is the exact model of the display you are using? Can you please provide some bootlogs?

There are also some pages with more info about the PDAs usage:
http://www.at91.com/linux4sam/bin/view/Linux4SAM/ConnectModuleFromPDA
http://www.at91.com/linux4sam/bin/view/Linux4SAM/PDADetectionAtBoot
http://www.at91.com/linux4sam/bin/view/Linux4SAM/SelectingPDAatBoot

Regards,
Razvan

TMScottSafety, 2019-06-25:
It is A PDA 4.3”
https://www.digikey.com/product-detail/en/precision-design-associates-inc/90-00123-A0/1230-1007-ND/6099992
Thank you.
Traian

http://www.at91.com/linux4sam/bin/view/Linux4SAM/ConnectModuleFromPDA
http://www.at91.com/linux4sam/bin/view/Linux4SAM/PDADetectionAtBoot
http://www.at91.com/linux4sam/bin/view/Linux4SAM/SelectingPDAatBoot

3M Notice: This communication is from an [EXTERNAL] sender.
If this email looks suspicious, do NOT click or open any links or attachments in the email. To report a suspicious email, click on the Report Phish – Phish Alert icon in the Outlook ribbon or forward this email using the report email as spam link in the text below.

Click here<https://spam.mmm.com/pem/pages/digestProcess/digestProcess.jsf?content=aedaaa864ecbae94fd60c3e824860ed94c7377a9172bbf106303ae966b66d61b7f84f37f885629e2d9f11863505cd596c3c50e625afd458a14727c3fcfdaa5703c98421c748e653f0e88611e8b53aae113732ef272febcc05f5db7fc40b8ee047cf6eec2862bb6ad2facef3ff87e4959c09ad56c9b15e8045225e4662ec315425a7e77cc50eb7182> to report this email as spam

RazvanStef, 2019-06-25:
Hello,

I just tested on SAMA5D27 SOM1-ek  with a TM4301B display, as you suggested. Display works fine with the image downloaded from http://www.at91.com/linux4sam/bin/view/Linux4SAM/Sama5d27Som1EKMainPage#Demo_archives.

Can you see if the display is detected in u-boot?

Regards,
Razvan

TMScottSafety, 2019-06-25:
Hi Razvan,
The display is blank…
Here is the login Tera Term print out:

RomBOOT

AT91Bootstrap 3.8.11 (Fri Oct 12 06:34:28 EEST 2018)

SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SD: Card Capacity: High or Extended
SD: Specification Version 3.0X
SD/MMC: Done to load image
<debug_uart>

U-Boot 2018.07-linux4sam_6.0 (Oct 12 2018 - 06:34:36 +0300)

CPU: SAMA5D27 1G bits DDR2 SDRAM
Crystal frequency:       24 MHz
CPU clock        :      492 MHz
Master clock     :      164 MHz
DRAM:  128 MiB
MMC:   sdio-host@a0000000: 0, sdio-host@b0000000: 1
Loading Environment from FAT... OK
In:    serial@f8020000
Out:   serial@f8020000
Err:   serial@f8020000
Net:   eth0: ethernet@f8008000
Hit any key to stop autoboot:  0
3952756 bytes read in 247 ms (15.3 MiB/s)
## Loading kernel from FIT Image at 21000000 ...
   Using 'kernel_dtb' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux4SAM Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x210000e0
     Data Size:    3887448 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x22000000
     Entry Point:  0x22000000
     Hash algo:    crc32
     Hash value:   284556d8
     Hash algo:    sha1
     Hash value:   21a2bc6b91b88b6b93507bda6efd0f0653911201
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 21000000 ...
   Using 'kernel_dtb' configuration
   Trying 'base_fdt' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x213b5364
     Data Size:    44915 Bytes = 43.9 KiB
     Architecture: ARM
     Load Address: 0x23000000
     Hash algo:    crc32
     Hash value:   44aa981a
     Hash algo:    sha1
     Hash value:   96610a3de96b76f7669da56de84f1aac5689c2f4
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Loading fdt from 0x213b5364 to 0x23000000
   Booting using the fdt blob at 0x23000000
   Loading Kernel Image ... OK
   Loading Device Tree to 27b47000, end 27b54f72 ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
[... 214 lines cut ...]
atmel_usba_udc 300000.gadget: FIFO at [mem 0x00300000-0x003fffff] mapped at c8d00000
OF: graph: no port node found in /ahb/apb/isc@f0008000
atmel_isc f0008000.isc: no subdev found
done
Initializing random number generator... random: dd: uninitialized urandom read (512 bytes read)
done.
Starting system message bus: random: dbus-uuidgen: uninitialized urandom read (12 bytes read)
random: dbus-uuidgen: uninitialized urandom read (8 bytes read)
done
Starting network: IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
udhcpc: started, v1.27.2
udhcpc: sending discover
udhcpc: sending discover
udhcpc: sending discover
udhcpc: no lease, failing
FAIL
Starting dropbear sshd: urandom_read: 2 callbacks suppressed
random: dropbear: uninitialized urandom read (32 bytes read)
OK
random: ssh-keygen: uninitialized urandom read (32 bytes read)
Starting sshd: OK
Starting Application Launcher: OK

Welcome to the Microchip Demo
sama5 login:

TMScottSafety, 2019-06-25:
Hi Razvan,
I am using the linux4sam-poky-sama5d27_som1_ek_video-6.0.img
It cannot see the screen, see below:
Any idea where I should look?
Is the error related to a dead display?
Thank you,
Traian

“
…..
* Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon
random: avahi-daemon: uninitialized urandom read (4 bytes read)
random: avahi-daemon: uninitialized urandom read (4 bytes read)          [ ok ]
Starting crond: OK
cat: '/sys/class/drm/card0/*/modes': No such file or directory

Starting Demo...
OK

Poky (Yocto Project Reference Distro) 2.5.1 sama5d27-som1-ek-sd /dev/ttyS0

sama5d27-som1-ek-sd login: random: ApplicationLaun: uninitialized urandom read (4 bytes read)
Could not open DRM device /dev/dri/card0 (No such file or directory)
linuxfb: Failed to initialize screen

Poky (Yocto Project Reference Distro) 2.5.1 sama5d27-som1-ek-sd /dev/ttyS0
“

RazvanStef, 2019-06-26:
Hello,
I tested the same image (ftp://www.at91.com/pub/demo/linux4sam_6.0/linux4sam-poky-sama5d27_som1_ek_video-6.0.img
[linux4sam-poky-sama5d27_som1_ek_video-6.0.img-bootlog.txt](https://github.com/linux4sam/meta-atmel/files/3328544/linux4sam-poky-sama5d27_som1_ek_video-6.0.img-bootlog.txt)
.bz2). From the logs you sent, you boot a buildroot image and there is no display detected at boot.

If u-boot detects the PDA, it should update the kernel bootargs as this:

> Kernel command line: console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait atmel.pm_modes=standby,ulp1 video=Unknown-1:800x480-16

I've attached the bootlog for linux4sam-poky-sama5d27_som1_ek_video-6.0.img image.

Can you try with another screen? 

Regards,
Razvan

ehristev, 2019-06-26:
Hello @TMScottSafety :
I am working on U-boot project and designed the detection.
Can you please stop the booting process in u-boot (when u-boot prints press any key to stop auto-boot).
and then issue 'printenv'
and let me know what is printed.
We can work out from there what is wrong with your setup.

As a side note, please be aware that U-boot automatically detects a screen and sets up for TM7000, TM7000B, TM5000 and TM4300.  If you screen is another flavor, but still compliant with detection, it will be detected and the 'pda' variable set accordingly. If your screen is not compliant with the detection then no detection will happen.
PDAs of 4.3" can be TM4300, TM4301, TM4301B etc.

TMScottSafety, 2019-06-26:
Hi Eugen,
I have a display model TM4301B from  www.pdaatl.com<http://www.pdaatl.com>

Here is the ‘printenv’ output:

RomBOOT

AT91Bootstrap 3.8.11 (Thu Oct 11 17:16:10 UTC 2018)

SD/MMC: Image: Read file u-boot.bin to 0x23f00000
MMC: ADMA supported
SD: Card Capacity: High or Extended
SD: Specification Version 3.0X
SD/MMC: Done to load image
<debug_uart>

U-Boot 2018.07-linux4sam_6.0 (Oct 11 2018 - 17:16:11 +0000)

CPU: SAMA5D27 1G bits DDR2 SDRAM
Crystal frequency:       24 MHz
CPU clock        :      492 MHz
Master clock     :      164 MHz
DRAM:  128 MiB
MMC:   sdio-host@a0000000: 0, sdio-host@b0000000: 1
Loading Environment from FAT... OK
In:    serial@f8020000
Out:   serial@f8020000
Err:   serial@f8020000
Net:   eth0: ethernet@f8008000
Hit any key to stop autoboot:  0
=> printenv<INTERRUPT>
=> printenv
at91_pda_detect=run pda4300test; run pda7000test; run pda7000btest; run pda5000test;
at91_prepare_bootargs=test -n $display_var && setenv bootargs ${bootargs} ${at91_video_bootargs}
at91_prepare_overlays_config=test -n $display_var && setenv at91_overlays_config '#'${display_var}
at91_prepare_video_bootargs=test -n $display_var && setenv at91_video_bootargs video=${video_mode}
at91_set_display=test -n $pda && setenv display $pda
bootargs=console=ttyS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 rootwait atmel.pm_modes=standby,ulp1
bootcmd=run at91_set_display; run at91_pda_detect; run at91_prepare_video_bootargs; run at91_prepare_bootargs; run at91_prepare_overlays_config; run bootcmd_boot;
bootcmd_boot=fatload mmc 0:1 0x24000000 sama5d27_som1_ek.itb; bootm 0x24000000#kernel_dtb${at91_overlays_config}
bootdelay=1
ethact=gmac0
ethaddr=80:1f:12:41:73:ca
fdtcontroladdr=27b56488
pda=4301B
pda4300test=test -n $display && test $display = 4300 && setenv display_var 'pda4' && setenv video_mode ${video_mode_pda4}
pda5000test=test -n $display && test $display = 5000 && setenv display_var 'pda5' && setenv video_mode ${video_mode_pda5}
pda7000btest=test -n $display && test $display = 7000B && setenv display_var 'pda7b' && setenv video_mode ${video_mode_pda7b}
pda7000test=test -n $display && test $display = 7000 && setenv display_var 'pda7' && setenv video_mode ${video_mode_pda7}
stderr=serial@f8020000
stdin=serial@f8020000
stdout=serial@f8020000
video_mode_pda4=Unknown-1:480x272-16
video_mode_pda5=Unknown-1:800x480-16
video_mode_pda7=Unknown-1:800x480-16
video_mode_pda7b=Unknown-1:800x480-16

Environment size: 1658/16380 bytes
=>

ehristev, 2019-06-26:
I see you have a special flavor of the 4300 PDA.

U-boot only checks for the 4300 display.
You can alter this configuration like this
=> edit pda4300test

then edit the line and make it look like this : 
test -n $display && test $display = 4301B && setenv display_var 'pda4' && setenv video_mode ${video_mode_pda4}

and press enter
and then 
=> saveenv
to save your changes

Notice that the only change is the replacement of '4300' with '4301B' (exactly the value of 'pda' , which is your detected screen ).

Let me know if this works

noglitch, 2019-06-26:
One information displayed here: http://www.at91.com/linux4sam/bin/view/Linux4SAM/SelectingPDAatBoot#Practical_example_with_TM4301B_s

TMScottSafety, 2019-06-26:
Thank you,
That is what I needed,
Traian

closed 2019-06-27 by RazvanStef