linux4sam/meta-atmel #140: warrior branch

texierp, 2019-06-25:
Hi @RazvanStef,

Are there any plans to release a warrior branch ?
It would be nice :) 

BR

/Pierre-Jean

RazvanStef, 2019-06-26:
Hello @texierp ,

We are currently working on a Linux4SAM release based on thud branch. After that is done, we will continue with a warrior branch.

Master branch has support for warrior. if you like to try it.

Best regards,
Razvan

texierp, 2019-06-26:
Hello,

> We are currently working on a Linux4SAM release based on thud branch. After that is done, we will continue with a warrior branch.

Ok, I understand, thanks for the clarification.

> Master branch has support for warrior. if you like to try it.

Already tested in fact, it's just to have the same branches on my repo, but I can wait, for sure ;).

BR
/Pierre-Jean

RazvanStef, 2019-06-26:
I can add a warrior branch to the repo, for development, if it makes it easier for you. But testing on it is going to be minimal for the moment.

Regards,
Razvan

closed 2019-06-26 by RazvanStef

texierp, 2019-06-26:
Many thanks @RazvanStef