linux4sam/meta-atmel #142: Qt5 doesn't build with warrior branch

andy-sdc, 2019-07-22:
The warrior branch of meta-atmel doesn't patch and build correctly against Qt 5.12.3.

The patches in meta-atmel/dynamic-layers/qt5-layer/recipes-qt/qt5/files need to be updated to match those in the thud branch.

-Andy.

RazvanStef, 2019-07-22:
Hello,

We are working on this. 

Regards,
Razvan

andy-sdc, 2019-07-22:
Hi Razvan,

Thanks for letting me know.  I have taken the files from the thud branch and replaced the ones in my working directory and QtBase now builds correctly.

I have found another build issue with QtWebEngine.  Looking in the log files directory it seems as though the only error I can see is in log.do_configure:

khronos development headers appear to be missing (mesa/libegl1-mesa-dev)
QtWebEngine will not be built.

Is this there another patch I need to cherrypick from thud to fix this?

-Andy.

RazvanStef, 2019-07-22:
We didn't investigated the WebEngine issues so there is no patch.

closed 2019-08-07 by RazvanStef