linux4sam/meta-atmel #148: Convert flexcom0 to i2c

agilesai1294, 2019-09-19:
Hi,

I am trying to connect ATECC608A to my SAMA5D2 xplained board. I have followed the steps from https://raw.githubusercontent.com/linux4sam/meta-atmel/sumo/README and build the image. The image was built successfully but i am having trouble connect ATECC608A to EXT1. 

When i do i2cdetect -y 0 i get the at91_i2c fc018600.i2c: controller timed out.

alexandrebelloni, 2019-09-23:
I would think that bus 0 is actually i2c0 and not the flexcom. How did you modify your device tree?

Anyway, I don't think this is the correct forum to discuss that as this is not directly related to meta-atmel.

noglitch, 2019-09-24:
Please discuss this on at91.com forum.
Best regards

closed 2019-09-24 by noglitch