linux4sam/meta-atmel #151: Qtbase version desyncronizaton for 'thud' branch

andrewtimoshenko, 2019-09-27:
Hi all,

There's desynchronization between **'thud'** branches of **'meta-atmel'** and **'meta-qt5'** layers.
I use **meta-qt5** layer - **commit 1520d5b2b2beec5e1c3209d3178219e93ef08bca** (HEAD -> thud, origin/thud-next, origin/thud, origin/jansa/thud),
where's used qtbase of 5.11.3 version.

But in the **meta-atmel** layer - commit **5d8d83e656246c8ca534c6288bbb7fbb7afda683** (HEAD -> thud, tag: linux4sam_6.1-rc5, tag: linux4sam_6.1, origin/thud)
is used .bbappend file with a patch for **qtbase 5.12.2** (0004-Provide-access-to-linuxfb-dri-fd-through-platform.patch):
> commit cbda4ed6c60cded44fe0895a4463ab652298403f
> Author: Razvan Stefanescu <razvan.stefanescu@microchip.com>
> Date:   Fri May 17 12:54:18 2019 +0300
> 
>     qtbase: Fix compilation issue for QT5.12.2
>     
>     Signed-off-by: Razvan Stefanescu <razvan.stefanescu@microchip.com>

Bitbake log:

> Build Configuration:
> BB_VERSION           = "1.40.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "universal"
> TARGET_SYS           = "arm-poky-linux-gnueabi"
> MACHINE              = "sama5d27-som1-ek-sd"
> DISTRO               = "poky-atmel"
> DISTRO_VERSION       = "2.6.2"
> TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard cortexa5"
> TARGET_FPU           = "hard"
> meta                 
> meta-poky            
> meta-yocto-bsp       = "HEAD:2dd610cc27c606f731f83047b4766787aab6e6b2"
> meta-oe              
> meta-python          
> meta-networking      = "HEAD:9b3b907f30b0d5b92d58c7e68289184fda733d3e"
> meta-mender-atmel    = "HEAD:efc8ae3c460f93f8ecbce1f76024a0a266e2992e"
> meta-mender-core     
> meta-mender-demo     = "thud:352cd81fe4dd6e44ba0819da1d8e189109e805e6"
> meta-iot-cloud       = "feature/bsp_porting_RLPR-64:7b0765f55fc5aa1dc55e15e415145e5d2c3c345f"
> meta-qt5             = "thud:1520d5b2b2beec5e1c3209d3178219e93ef08bca"
> meta-atmel           = "thud:5d8d83e656246c8ca534c6288bbb7fbb7afda683"
> meta-rollo           = "feature/bsp_porting_RLPR-64:7b0765f55fc5aa1dc55e15e415145e5d2c3c345f"
> 
> Initialising tasks: 100% |###############################################################################| Time: 0:00:01
> Sstate summary: Wanted 477 Found 12 Missed 465 Current 240 (2% match, 35% complete)
> NOTE: Executing SetScene Tasks
> NOTE: Executing RunQueue Tasks
> ERROR: qtbase-5.11.3+gitAUTOINC+08de243eaa-r0 do_patch: Command Error: 'quilt --quiltrc /home/timandr/projects/rollo/rollo-firmware/build_sama5d27/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/qtbase/5.11.3+gitAUTOINC+08de243eaa-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
> Applying patch 0004-Provide-access-to-linuxfb-dri-fd-through-platform.patch
> patching file src/platformsupport/fbconvenience/qfbscreen_p.h
> patching file src/plugins/platforms/linuxfb/qlinuxfbdrmscreen.cpp
> patching file src/plugins/platforms/linuxfb/qlinuxfbdrmscreen.h
> patching file src/plugins/platforms/linuxfb/qlinuxfbintegration.cpp
> Hunk #1 FAILED at 173.
> 1 out of 1 hunk FAILED -- rejects in file src/plugins/platforms/linuxfb/qlinuxfbintegration.cpp
> patching file src/plugins/platforms/linuxfb/qlinuxfbscreen.h
> Patch 0004-Provide-access-to-linuxfb-dri-fd-through-platform.patch does not apply (enforce with -f)
> ERROR: qtbase-5.11.3+gitAUTOINC+08de243eaa-r0 do_patch: Function failed: patch_do_patch
> ERROR: Logfile of failure stored in: /home/timandr/projects/rollo/rollo-firmware/build_sama5d27/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/qtbase/5.11.3+gitAUTOINC+08de243eaa-r0/temp/log.do_patch.9063
> ERROR: Task (/home/timandr/projects/rollo/rollo-firmware/build_sama5d27/../meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_patch) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 1672 tasks of which 673 didn't need to be rerun and 1 failed.
> 
> Summary: 1 task failed:
>   /home/timandr/projects/rollo/rollo-firmware/build_sama5d27/../meta-qt5/recipes-qt/qt5/qtbase_git.bb:do_patch
> Summary: There was 1 WARNING message shown.
> Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

Is there currently a solution to this issue?

Thanks.

texierp, 2019-09-28:
Hi @andrewtimoshenko,

Have you tried to revert this commit (git revert -s fa2bb29b2b057b86fbef75d9b1238042df8921cd). This to know if the build is fine without this patch.

andrewtimoshenko, 2019-09-30:
> Have you tried to revert this commit (git revert -s [fa2bb29](https://github.com/linux4sam/meta-atmel/commit/fa2bb29b2b057b86fbef75d9b1238042df8921cd)). This to know if the build is fine without this patch.

Yes, `git revert -s fa2bb29` works fine. 

Qtbase is successfully built.

Thanks!

texierp, 2019-09-30:
great, thanks @andrewtimoshenko for reporting back.

A pull request is now open to fix this issue (see https://github.com/linux4sam/meta-atmel/pull/152)

codrin989, 2020-01-28:
As the README of the 'thud' branch mentions, please use branch 5.12 for meta-qt5 layer:
https://github.com/linux4sam/meta-atmel/blob/thud/README#L44

closed 2020-01-28 by codrin989