linux4sam/meta-atmel #160: meta-atmel support for at91sam9m10g45ek

fberndl, 2020-01-09:
Is meta-atmel and the other linux4sam repositories still supporting at91sam9m10g45ek in version zeus? 

Should the following configuration work out of the box with the evaluation kit?
meta-atmel --> zeus
linux --> 4.19
u-boot --> 2019.04
at91bootstrap --> master

Thanks

codrin989, 2020-01-09:
Unfortunately, at91sam9m10g45ek is no longer part of our official releases. However, we are always trying to keep backward compatibility and not break older boards, so it might work with minor to none changes. Just try it, if you can, and see how it works. You can always make a pull request to our linux4sam repositories, to fix things that you need.

fberndl, 2020-01-21:
Thanks, can be closed

closed 2020-01-21 by fberndl