linux4sam/meta-atmel #161: Is angstrom-sdk and arm sdk necessary for at91sam9m10 compilation?

fberndl, 2020-01-13:
Hi, 

in older versions of u-boot the angstrom-sdk ( for compiling u-boot) and arm sdk ( for compiling bootstrap) was needed if I understood it correctly. 

I did not find any depenency to this sdk's so they are not needed anymore using current yocto releases?

SOC: at91sam9m10

codrin989, 2020-01-28:
I don't know about angstrom or arm sdks, but you can find the dependencies by typing 'bitbake -g <packet_name>' and investigating the recipe-depends.dot file. Do you have issues compiling u-boot/at91bootstrap for at91sam9m10?

fberndl, 2020-01-28:
Hi,

it seems that angstrom and arm sdks are quite old. before yocto/poky had in-built support of it. So it should not be needed anymore.

closed 2020-01-28 by fberndl