linux4sam/meta-atmel #172: bitbake cryptoauthlib doesnt found the recipe?

hsnhrn, 2020-07-09:
Hi i have cloned the layer and trying to build it using bitbake. But the recipes in dynamic-layers are not being found with bitbake command. How can i add recipes which are in dynamic-layers folder into my image?

codrin989, 2020-07-09:
Hi @hsnhrn .  For the recipes from dynamic-layers/aws-layer you need to also clone meta-aws layer, from https://github.com/aws/meta-aws . Please see https://github.com/linux4sam/meta-atmel/blob/dunfell/README#L45 for more details. Are you trying to build only cryptoauthlib?

hsnhrn, 2020-07-09:
hi @codrin989 thanks for your reply,  i also need python3-cryptoauthlib..

hsnhrn, 2020-07-10:
I cloned the meta-aws and added it to my bblaers but getting this error on ruunning bitbake cryptoauthlib
ERROR: Nothing PROVIDES 'eudev' (but /home/hassan_dondon/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-atmel/dynamic-layers/aws-layer/recipes-security/cryptoauthlib/cryptoauthlib_git.bb DEPENDS on or otherwise requires it)
eudev was skipped: conflicting distro feature 'systemd' (in DISTRO_FEATURES)
ERROR: Required build target 'cryptoauthlib' has no buildable providers.
Missing or unbuildable dependency chain was: ['cryptoauthlib', 'eudev']

hsnhrn, 2020-07-10:
I cloned the meta-aws and added it to my bblaers but getting this error on ruunning bitbake cryptoauthlib
ERROR: Nothing PROVIDES 'eudev' (but /home/user/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-atmel/dynamic-layers/aws-layer/recipes-security/cryptoauthlib/cryptoauthlib_git.bb DEPENDS on or otherwise requires it)
eudev was skipped: conflicting distro feature 'systemd' (in DISTRO_FEATURES)
ERROR: Required build target 'cryptoauthlib' has no buildable providers.
Missing or unbuildable dependency chain was: ['cryptoauthlib', 'eudev']

codrin989, 2020-07-10:
It looks like there is a conflict between systemd and eudev. If you really want to use systemd instead of sysvinit, try replacing 'eudev' with 'udev' here: https://github.com/linux4sam/meta-atmel/blob/dunfell/dynamic-layers/aws-layer/recipes-security/cryptoauthlib/cryptoauthlib_git.bb#L11

hsnhrn, 2020-07-10:
Thanks a lot. worked fine now!!

colin-pm, 2020-07-15:
@codrin989 I know this task is closed, but can I ask why meta-aws is a dependency of cryptoauthlib?

codrin989, 2020-07-15:
Hi @colin-pm . For us, cryptoauthlib only made sense when used in relation with greengrass. If this is not the case (and it looks like it's not), please make a new feature request. Or you could make a pull request directly :)

hsnhrn, 2020-07-16:
@codrin989  i am trying to run these examples https://github.com/MicrochipTech/cryptoauthtools/tree/master/python/examples
but i am not able to import cryptoauthlib and got this error below :
File "/usr/lib/python3.8/site-packages/cryptoauthlib/__init__.py". line 16. in <module> load_cryptoauthlib()
File "/usr/lib/python3.8/site-packages/cryptoauthlib/library.py". line 83. in  load_cryptoauthlib()
_CRYPTO_LIB =ctypes.cdll.LoadLibrary(os.path.join(curr_path. "libcryptoauth.so"))
File "/usr/lib/python3.8/ctypes/__init__.py". line 451. in  LoadLibrary return self._dlltype(name)
File "/usr/lib/python3.8/ctypes/__init__.py". line 373. in  __init__ self._handle = _dlopen(self._name mode)
OSError : /usr/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.so: undefined symbol: hid_open

 
As you mentioned above it only make sense to use it with greengrass. So is this error related to soemthing like that? 
and can i fix this somehow?

codrin989, 2020-07-16:
Hi @hsnhrn . Yes, please open a new post, this is something different. Thanks!

closed 2020-07-16 by hsnhrn