linux4sam/meta-atmel #174: Import error cryptoauthlib.

hsnhrn, 2020-07-16:
i am trying to run these examples https://github.com/MicrochipTech/cryptoauthtools/tree/master/python/examples
but i am not able to import cryptoauthlib and got this error below :
File "/usr/lib/python3.8/site-packages/cryptoauthlib/init.py". line 16. in load_cryptoauthlib()
File "/usr/lib/python3.8/site-packages/cryptoauthlib/library.py". line 83. in load_cryptoauthlib()
_CRYPTO_LIB =ctypes.cdll.LoadLibrary(os.path.join(curr_path. "libcryptoauth.so"))
File "/usr/lib/python3.8/ctypes/init.py". line 451. in LoadLibrary return self._dlltype(name)
File "/usr/lib/python3.8/ctypes/init.py". line 373. in init self._handle = _dlopen(self._name mode)
OSError : /usr/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.so: undefined symbol: hid_open

How can i fix this? thanks a lot!

codrin989, 2020-07-16:
Please try https://github.com/MicrochipTech/cryptoauthlib/issues/136 .

hsnhrn, 2020-07-16:
i have a yocto os and i added the recipe  in the project and then bitbake myimage. how can i install without cache in yocto?

codrin989, 2020-07-16:
This issue seems to be related mainly to cryptoauthlib itself, which is handled by a different team. I see there are many similar issues reported:
https://github.com/MicrochipTech/cryptoauthtools/issues/16
https://github.com/MicrochipTech/cryptoauthlib/issues/144
+ the one I linked above
Please look them up.

hsnhrn, 2020-07-17:
Seems like a version problem.  Can i somehow change the version in the recipe?

codrin989, 2020-07-17:
You have rename python3-cryptoauthlib_20200208.bb  recipe to python3-cryptoauthlib_<your_version>.bb .

hsnhrn, 2020-07-17:
ERROR: python3-cryptoauthlib-20190517-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1198"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/sysroots-uninative/x86_64-linux/usr/bin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/bin/python3-native:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/openembedded-core/scripts:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot/usr/bin/crossscripts:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/sbin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/bin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/sbin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/bin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/openembedded-core/bitbake/bin:/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/hosttools"; export HOME="/home/user"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -P /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/downloads 'https://files.pythonhosted.org/packages/1b/ed/da1709095abd203e37892f7183b68433382ce6a0f6129dfa0ac47e2ba85c/cryptoauthlib-20190517.tar.gz' --progress=dot -v failed with exit code 8, output:
--2020-07-17 10:25:28--  https://files.pythonhosted.org/packages/1b/ed/da1709095abd203e37892f7183b68433382ce6a0f6129dfa0ac47e2ba85c/cryptoauthlib-20190517.tar.gz
Resolving files.pythonhosted.org (files.pythonhosted.org)... 151.101.113.63, 2a04:4e42:3::319
Connecting to files.pythonhosted.org (files.pythonhosted.org)|151.101.113.63|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-07-17 10:25:28 ERROR 404: Not Found.

ERROR: python3-cryptoauthlib-20190517-r0 do_fetch: Fetcher failure for URL: 'https://files.pythonhosted.org/packages/1b/ed/da1709095abd203e37892f7183b68433382ce6a0f6129dfa0ac47e2ba85c/cryptoauthlib-20190517.tar.gz'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/log.do_fetch.3125
ERROR: Task (/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/layers/meta-atmel/dynamic-layers/aws-layer/recipes-support/python-cryptoauthlib/python3-cryptoauthlib_20190517.bb:do_fetch) failed with exit code '1'

I am trying this version
https://pypi.org/project/cryptoauthlib/20190517/

codrin989, 2020-07-17:
Right, you also have to update SRC_URI and replace https://github.com/linux4sam/meta-atmel/blob/dunfell/recipes-support/python-cryptoauthlib/python3-cryptoauthlib_20200208.bb#L7 with 
SRC_URI = "https://files.pythonhosted.org/packages/e3/b0/c7377a71078ca77cc8615d5bb90e717ef541bda8becff251cd557eabe641/cryptoauthlib-${PV}.tar.gz ${PYPI_SRC_URI}"

hsnhrn, 2020-07-17:
checksum mismatch error now. from where can i find the checksum of that version?

SRC_URI[md5sum] = "3a464cb6ea78286353870e544f85e208"
SRC_URI[sha256sum] = "7b04a4097c6f8d4b539c7425261eac3353016405b863fec2d827152ce3a652bf"

codrin989, 2020-07-17:
The easiest way is to look at the error message reported by yocto and replace the checksums in the recipe with the ones yocto expects to find.

hsnhrn, 2020-07-17:
Now have this error.
ERROR: python3-cryptoauthlib-20190517-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: python3-cryptoauthlib-20190517-r0 do_compile: Execution of '/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/run.do_compile.4098' failed with exit code 1:
/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py:42: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.platform is 'win32':
WARNING: The pip package is not available, falling back to EasyInstall for handling setup_requires/test_requires; this is deprecated and will be removed in a future version.
ERROR: Do not try to fetch `wheel' for building. Please add its native recipe to DEPENDS.
Traceback (most recent call last):
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py", line 186, in <module>
    setup(
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 143, in setup
    _install_setup_requires(attrs)
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 138, in _install_setup_requires
    dist.fetch_build_eggs(dist.setup_requires)
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/dist.py", line 718, in fetch_build_eggs
    resolved_dists = pkg_resources.working_set.resolve(
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'wheel' distribution was not found and is required by the application
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/log.do_compile.4098

hsnhrn, 2020-07-17:
https://github.com/MicrochipTech/cryptoauthlib/issues/176 
I got this response from above..

"This would be a configuration issue. Are you going to be using a kit such as the CryptoAuth Trust Platform Development Kit? If you are then you'll have to include USB support (e.g. libudev) to your build. If you are not using a USB kit then you'll need to set the environment variable CRYPTOAUTHLIB_NOUSB for the python build."

can i set it somehow in your recipe?  maybe the latest version will work this way and i dont need to downgrade the version?

codrin989, 2020-07-17:
try adding python3-wheel and udev in DEPENDS here https://github.com/linux4sam/meta-atmel/blob/dunfell/recipes-support/python-cryptoauthlib/python3-cryptoauthlib_20200208.bb#L17

hsnhrn, 2020-07-17:
i added it 
DEPENDS += "cmake-native python3-wheel udev" 
but still got the error

ERROR: python3-cryptoauthlib-20190517-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: python3-cryptoauthlib-20190517-r0 do_compile: Execution of '/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/run.do_compile.11635' failed with exit code 1:
/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py:42: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.platform is 'win32':
WARNING: The pip package is not available, falling back to EasyInstall for handling setup_requires/test_requires; this is deprecated and will be removed in a future version.
ERROR: Do not try to fetch `wheel' for building. Please add its native recipe to DEPENDS.
Traceback (most recent call last):
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py", line 186, in <module>
    setup(
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 143, in setup
    _install_setup_requires(attrs)
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 138, in _install_setup_requires
    dist.fetch_build_eggs(dist.setup_requires)
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/dist.py", line 718, in fetch_build_eggs
    resolved_dists = pkg_resources.working_set.resolve(
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'wheel' distribution was not found and is required by the application
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/log.do_compile.11635

codrin989, 2020-07-17:
Make it python3-wheel-native instead of python3-wheel. My mistake.

hsnhrn, 2020-07-17:
ERROR: python3-cryptoauthlib-20190517-r0 do_compile: 'python3 setup.py build ' execution failed.
ERROR: python3-cryptoauthlib-20190517-r0 do_compile: Execution of '/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/run.do_compile.17729' failed with exit code 1:
running build
running build_py
creating /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8
creating /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/tng.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/iface.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/atcacert.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/atjwt.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/device.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/library.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/__init__.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/atcaenum.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/status.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/exceptions.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
copying cryptoauthlib/atcab.py -> /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/build/lib.linux-x86_64-3.8/cryptoauthlib
running egg_info
writing cryptoauthlib.egg-info/PKG-INFO
writing dependency_links to cryptoauthlib.egg-info/dependency_links.txt
writing requirements to cryptoauthlib.egg-info/requires.txt
writing top-level names to cryptoauthlib.egg-info/top_level.txt
reading manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'cryptoauthlib.egg-info/SOURCES.txt'
running build_ext
/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py:42: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.platform is 'win32':
Traceback (most recent call last):
  File "/home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/cryptoauthlib-20190517/setup.py", line 148, in build_extension
    subprocess.check_output(['cmake', cmakelist_path] + cmake_args, cwd=os.path.abspath(self.build_temp), stderr=subprocess.STDOUT)
[... cut ...]
   variable CRYPTOAUTHLIB_NOUSB to true before trying to install
   this package: 

       $ export CRYPTOAUTHLIB_NOUSB=True

   Run setup.py clean before trying install again or use the pip 
   option --no-cache-dir

WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/user/STM32MP15-Ecosystem-v2.0.0/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-06-24/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/python3-cryptoauthlib/20190517-r0/temp/log.do_compile.17729

hsnhrn, 2020-07-17:
"$ export CRYPTOAUTHLIB_NOUSB=True" this is also mentioned by developer on this repo MicrochipTech/cryptoauthlib#176

hsnhrn, 2020-07-20:
That worked. I am able to import this version now. thanks a lot

closed 2020-07-20 by hsnhrn

codrin989, 2020-07-20:
Great! Could you please retry without changing the version of the packet? And add only CRYPTOAUTHLIB_NOUSB = "True" and udev in DEPENDS . If it works, could you please send a pull-request with these changes only? Thanks!

hsnhrn, 2020-07-20:
yeah, newer version worked too. i sent you a pull-request with that changes. Thanks a lot!