linux4sam/meta-atmel #188: I'm using a file system compiled from a recipe, and I get errors running egt-launche program.

blithe2015, 2020-12-02:
root@NECRO:~# egt-launcher 
terminate called after throwing an instance of 'std::runtime_error'
  what():  unable to open DRM driver
Aborted (core dumped)

Does this error occur because drm is not supported?

ldesroches, 2020-12-02:
Hi,

No it doesn't mean that DRM is not supported, it means that it can't open the driver so it's probably missing. Maybe the overlay for the screen is not loaded preventing the probe of the LCD driver.

Ludovic

noglitch, 2020-12-02:
Hi @blithe2015 , can we have more info about the platform? LCD used? Yocto Project image generated?

blithe2015, 2020-12-03:
> platform
The platform is STM32MP157. LCD is used,the image is Successful.

noglitch, 2020-12-03:
Probably components missing on this platform. Works well on all our SoCs. I close it.

closed 2020-12-03 by noglitch