linux4sam/meta-atmel #194: Error compiling "The postinstall intercept hook" for sama5d3-xplained

jarus0, 2020-12-09:
All steps followed as written in [README](https://github.com/linux4sam/meta-atmel/blob/dunfell/README)

Step 8 command for my board is given below
`MACHINE=sama5d3-xplained-sd bitbake core-image-minimal`

1) **Compiling error below given**
```
Initialising tasks: 100% |############################################################################################| Time: 0:00:03
Sstate summary: Wanted 114 Found 104 Missed 10 Current 946 (91% match, 99% complete)
NOTE: Executing Tasks
ERROR: core-image-minimal-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gio_module_cache' failed, details in /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs
ERROR: Logfile of failure stored in: /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs.29400
ERROR: Task (/home/dnk025/my_dir/poky/meta/recipes-core/images/core-image-minimal.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3266 tasks of which 3249 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dnk025/my_dir/poky/meta/recipes-core/images/core-image-minimal.bb:do_rootfs
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

**2) log.do_rootfs.29400** file contain given below
```
NOTE: Running ['oe-pkgdata-util', '-p', '/home/dnk025/my_dir/poky/build-microchip/tmp/pkgdata/sama5d3-xplained-sd', 'glob', '/tmp/installed-pkgsnqhmgoga', '']
NOTE: Installing complementary packages ...  (skipped already provided packages )
NOTE: Running intercept scripts:
NOTE: > Executing update_gio_module_cache intercept ...
NOTE: Exit code 1. Output:
+ [ True = False -a qemuwrapper-cross != nativesdk-qemuwrapper-cross ]
+ qemu-arm -r 3.2.0 -E LD_LIBRARY_PATH=/home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/rootfs/usr/lib:/home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/rootfs/lib -L /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/rootfs /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/rootfs/usr/libexec/gio-querymodules /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/rootfs/usr/lib/gio/modules/

ERROR: The postinstall intercept hook 'update_gio_module_cache' failed, details in /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs
DEBUG: Python function do_rootfs finished
```

**3) Note:- I don't have below directories in rootfs/usr**
```
/lib/gio/modules/
/libexec/gio-querymodules
```
How can i resolve this issue?
Thanks

codrin989, 2020-12-09:
I just rechecked and it seems fine on my side. My Build Configuration is:
$ MACHINE=sama5d3-xplained-sd bitbake  core-image-minimal
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:12
Parsing of 2281 .bb files complete (0 cached, 2281 parsed). 3382 targets, 361 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d3-xplained-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "3.1.4"
TUNE_FEATURES        = "arm vfp cortexa5 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:0d4f80535e7ebb799a70761e83a40905f2c14535"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-initramfs       = "dunfell:f2d02cb71eaff8eb285a1997b30be52486c160ae"
meta-atmel           = "dunfell:78b92b0973b540d98ed4fafdd15d82d66f0b9d58"
meta-multimedia      = "dunfell:f2d02cb71eaff8eb285a1997b30be52486c160ae"
meta-aws             = "dunfell:e34ed15a87e36fa4199c63a184630fa370abac55"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 718 Found 12 Missed 706 Current 342 (1% match, 33% complete)
NOTE: Executing Tasks
WARNING: core-image-minimal-1.0-r0 do_prepare_recipe_sysroot: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?
WARNING: core-image-minimal-1.0-r0 do_prepare_recipe_sysroot: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?
WARNING: core-image-minimal-1.0-r0 do_rootfs: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?
WARNING: core-image-minimal-1.0-r0 do_rootfs: [log_check] core-image-minimal: found 1 warning message in the logfile:
[log_check] WARNING: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?

WARNING: core-image-minimal-1.0-r0 do_image_wic: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?
WARNING: core-image-minimal-1.0-r0 do_image_complete: Manifest /home/codrin/yocto/poky/mchp/tmp/sstate-control/manifest-x86_64-nasm-native.populate_sysroot not found in x86_64 (variant '')?
NOTE: Tasks Summary: Attempted 3275 tasks of which 1454 didn't need to be rerun and all succeeded.

Summary: There were 6 WARNING messages shown.

Could you please check your Build Configuration and make sure it's similar? You might need to bump poky...

jarus0, 2020-12-10:
Still having error

```
Initialising tasks: 100% |############################################################################################| Time: 0:00:03
Sstate summary: Wanted 875 Found 0 Missed 875 Current 185 (0% match, 17% complete)
NOTE: Executing Tasks
ERROR: binutils-cross-arm-2.34-r0 do_unpack: Unpack failure for URL: 'git://sourceware.org/git/binutils-gdb.git;branch=binutils-2_34-branch;protocol=git'. No up to date source found: clone directory not available or not up to date: /home/dnk025/jyocta/poky/build-microchip/downloads/git2/sourceware.org.git.binutils-gdb.git; shallow clone not enabled
ERROR: Logfile of failure stored in: /home/dnk025/jyocta/poky/build-microchip/tmp/work/x86_64-linux/binutils-cross-arm/2.34-r0/temp/log.do_unpack.3059
ERROR: Task (/home/dnk025/jyocta/poky/meta/recipes-devtools/binutils/binutils-cross_2.34.bb:do_unpack) failed with exit code '1'
ERROR: glibc-2.31+gitAUTOINC+6fdf971c9d-r0 do_unpack: Unpack failure for URL: 'git://sourceware.org/git/glibc.git;branch=release/2.31/master;name=glibc'. No up to date source found: clone directory not available or not up to date: /home/dnk025/jyocta/poky/build-microchip/downloads/git2/sourceware.org.git.glibc.git; shallow clone not enabled
ERROR: Logfile of failure stored in: /home/dnk025/jyocta/poky/build-microchip/tmp/work/cortexa5t2hf-vfp-poky-linux-gnueabi/glibc/2.31+gitAUTOINC+6fdf971c9d-r0/temp/log.do_unpack.3060
ERROR: Task (/home/dnk025/jyocta/poky/meta/recipes-core/glibc/glibc_2.31.bb:do_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 639 tasks of which 637 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/dnk025/jyocta/poky/meta/recipes-devtools/binutils/binutils-cross_2.34.bb:do_unpack
  /home/dnk025/jyocta/poky/meta/recipes-core/glibc/glibc_2.31.bb:do_unpack
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

Below one thing I noticed
if I try to get the file using wget in terminal then it will show some error
`wget http://downloads.yoctoproject.org/mirror/sources/git2_sourceware.org.git.glibc.git.tar.gz` 
```
git2_sourceware.org.git.glibc.git   8%[===>                                                       ]  42.60M   233KB/s    in 3m 6s   

2020-12-10 07:14:53 (235 KB/s) - Connection closed at byte 44666976. Retrying.

--2020-12-10 07:14:54--  (try: 2)  http://downloads.yoctoproject.org/mirror/sources/git2_sourceware.org.git.glibc.git.tar.gz
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|198.145.29.63|:80... connected.
HTTP request sent, awaiting response... 416 Requested range not satisfiable

    The file is already fully retrieved; nothing to do.
```

but when I change protocol HTTP to HTTPS in wget URL then that's work fine.

jarus0, 2020-12-10:
Thanks for support
Issue resolve after upgrading my internet speed.

closed 2020-12-10 by jarus0