linux4sam/meta-atmel #196: Are the tools under scripts/ still relevant?

fedepell, 2020-12-13:
I am currently working with a sam9x20ek and was looking into having a simplified flashing procedure with what `meta-atmel` offers. I've been therefore looking to improve the scripts (shell and tcl) under `scripts/` for the board I'm using but after some studing my feeling is that the code there is totally obsolete now, as I would think that the way it is calling `sam-ba` is wrong (ie. all missing command options) and that now `sam-ba` automation is not TCL based anymore but uses QML (I cannot see the whole history of the `sam-ba` tool as in GitHub I can only got back as far as 3.0 or so).

Am I correct or did I really overlook something? If I'm correct I would suggest (I will gladly do a PR) to remove that scripts as they can be very confusing for newcomers and are obviously not working. If I'm wrong I'd appreciate if you can give me some pointers as I'd gladly improve it for the sam9x20ek I'm using.

Should my understanding that that code is obsolete, would replacing those scripts with something similar but QML based (as offered in the demo images already) be worth an inclusion into meta-atmel?

Thanks!
Federico

noglitch, 2021-06-09:
@fedepell ,
You analysis makes sense to me and indeed this script is not updated to cover our newer products nor SAM-BA 3.x.
I advice you to take the reference scripts that are included in .zip demo images: these are the ones we maintain. We are discussing internally in order to address this problem and see if we should remove this one.
Thanks for the heads-up. Best regards,
  Nicolas

codrin989, 2021-07-13:
Hi @fedepell . We just removed the scripts., since they are no longer supported.

closed 2021-07-13 by codrin989