linux4sam/meta-atmel #202: Secure Boot suppot for SAMA5D2

Noufalp90, 2021-04-14:
Hi Team,

Does the meta-atmel support secure boot on SAMA5D2?

Regards
Noufal P

codrin989, 2021-04-14:
Hi @Noufalp90 . There is an article here https://connect.ed-diamond.com/GNU-Linux-Magazine/GLMF-221/U-Boot-a-la-decouverte-du-demarrage-verifie made by @texierp . with some steps on how to enable secure boot on SAMA5D27-SOM1 with meta-atmel. Hope it helps.

Noufalp90, 2021-04-14:
It supports English? am getting it in another language. 
+ Is it possible to build image using yocto for secure boot?

codrin989, 2021-04-14:
meta-atmel does not generate secure boot images out-of-the-box. You could take a look at @texierp  's fork https://github.com/texierp/meta-atmel and see whether it includes any changes for secure boot.

Noufalp90, 2021-04-14:
Okay. So using buildtroot is required for secure boot support. Correct me  if am wrong.

codrin989, 2021-04-14:
buildroot is in the same situation as meta-atmel. The buildroot images are similar with the meta-atmel ones.

Noufalp90, 2021-04-14:
Okay. Even with yocto also, secure boot is possible?

codrin989, 2021-04-14:
secure boot is not restricted to a certain build system. I provided you above with a link to an article that explains very nicely how to do this with meta-atmel. Even if it's in french, I am sure you can find an easy way to translate it. There is also [this](http://ww1.microchip.com/downloads/en/AppNotes/DS00002748A.pdf) application note that uses buildroot and shows what changes are needed for secure boot. If you have other questions, please make a [support ticket](https://microchipsupport.force.com/s/). Thanks!

closed 2021-04-14 by codrin989