linux4sam/meta-atmel #207: sam-ba tarball download page missing

iancoolidge89, 2021-11-03:
ERROR: sam-ba-native-3.2.3-r0 do_fetch: Fetcher failure for URL: 'https://github.com/atmelcorp/sam-ba/releases/download/v3.2.3/sam-ba_3.2.3-linux_x86_64.tar.gz'. Unable to fetch URL from any source.

The entire https://github.com/atmelcorp/sam-ba/ is missing completely, this effects all the sam-ba recipes.

noglitch, 2021-11-03:
Fix proposed here: https://github.com/linux4sam/meta-atmel/pull/206
We'll address this issue in the next few days. Thanks for the feedback.
  Nicolas

hilt0n, 2021-11-08:
Very bad migration seriously or a very bad mistake...

alexandrebelloni, 2021-11-08:
I'm not sure how this comments helps but if this impacts you directly, you should think a bit more about how you are depending on third parties to build your images and what is you plan when those third parties disappear.

hilt0n, 2021-11-08:
This comment was not intended to Linux4SAM but for for Atmel/Microchip and it's indeed not really helpful since I doubt it will change anything in the OSS management of Atmel/Microchip. 
Your remark is like every project should contains the worldwide web, sorry but I don't agree. We use and trust meta-atmel because we buy products from them and we though it would be managed the right way, with a archived project for example, that's the way to do it.

noglitch, 2021-11-08:
Fix is taking more time than expecting because involving several teams. We're on it and pay attention to our user, be sure about that.
My advice is to remove sam-ba tool fetch from YP build as it's not completely related to the building a rootfs/kernel/bootloaders but I understand it's not the short term solution that you're expecting.
Regards,
  Nicolas

hilt0n, 2021-11-08:
We have fixed it by adding `EXTRA_IMAGEDEPENDS_remove = "sam-ba-native"` in our meta-atmel append meta.

I didn't want to be rude but I think it's also important to notify there is some real companies using this project and that we make some assumptions like stable YP fetching sources. I appreciate your feedback then.

heavyweight87, 2021-11-09:
@noglitch Will new versions also be added to github repo?

noglitch, 2021-11-09:
@heavyweight87 : I cannot tell. But I know that newer versions of sam-ba already released are not in this github repo, so I wouldn't count too much on that.

closed 2021-11-09 by noglitch