linux4sam/meta-atmel #209: populate_sdk build failed

HiFiPhile, 2021-12-17:
I've followed the [build procedure](https://www.linux4sam.org/bin/view/Linux4SAM/PokyBuild#Atmel_Poky_SDK) in order to build an image and sdk for `sama5d27-wlsom1-ek` on Debian 11 & Ubuntu 21.04 host with the same result.

While `bitbake microchip-headless-image` successfully built, `bitbake -c populate_sdk microchip-headless-image` failed with a locale error.
```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d27-wlsom1-ek-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "3.1.13"
TUNE_FEATURES        = "arm vfp cortexa5 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "dunfell:795339092f87672e4f68e4d3bc4cfd0e252d1831"
meta-oe
meta-networking
meta-webserver
meta-python
meta-initramfs       = "dunfell:69f94af4d91215e7d4e225bab54bf3bcfee42f1c"
meta-atmel           = "dunfell:20eeec4910f1214b9099f5276a944e5d281b70ee"
meta-multimedia      = "dunfell:69f94af4d91215e7d4e225bab54bf3bcfee42f1c"

Initialising tasks: 100% |################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 1341 Found 0 Missed 1341 Current 395 (0% match, 22% complete)
NOTE: Executing Tasks
ERROR: microchip-headless-image-1.0-r1 do_populate_sdk: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_populate_sdk(d)
     0003:
File: '/P4510/sama5d2/poky/meta/classes/populate_sdk_base.bbclass', lineno: 172, function: do_populate_sdk
     0168:
     0169:    populate_sdk(d)
     0170:
     0171:fakeroot python do_populate_sdk() {
 *** 0172:    populate_sdk_common(d)
     0173:}
     0174:SSTATETASKS += "do_populate_sdk"
     0175:SSTATE_SKIP_CREATION_task-populate-sdk = '1'
     0176:do_populate_sdk[cleandirs] = "${SDKDEPLOYDIR}"
File: '/P4510/sama5d2/poky/meta/classes/populate_sdk_base.bbclass', lineno: 169, function: populate_sdk_common
     0165:                    manifest_type=Manifest.MANIFEST_TYPE_SDK_HOST)
     0166:    create_manifest(d, manifest_dir=d.getVar('SDK_DIR'),
     0167:                    manifest_type=Manifest.MANIFEST_TYPE_SDK_TARGET)
     0168:
 *** 0169:    populate_sdk(d)
     0170:
     0171:fakeroot python do_populate_sdk() {
     0172:    populate_sdk_common(d)
     0173:}
File: '/P4510/sama5d2/poky/meta/lib/oe/sdk.py', lineno: 415, function: populate_sdk
     0411:    img_type = d.getVar('IMAGE_PKGTYPE')
     0412:    if img_type == "rpm":
     0413:        RpmSdk(d, manifest_dir).populate()
     0414:    elif img_type == "ipk":
 *** 0415:        OpkgSdk(d, manifest_dir).populate()
     0416:    elif img_type == "deb":
     0417:        DpkgSdk(d, manifest_dir).populate()
     0418:
     0419:    os.environ.clear()
File: '/P4510/sama5d2/poky/meta/lib/oe/sdk.py', lineno: 42, function: populate
     0038:    def populate(self):
     0039:        self.mkdirhier(self.sdk_output)
     0040:
     0041:        # call backend dependent implementation
 *** 0042:        self._populate()
     0043:
     0044:        # Don't ship any libGL in the SDK
     0045:        self.remove(os.path.join(self.sdk_output, self.sdk_native_path,
     0046:                         self.d.getVar('libdir_nativesdk').strip('/'),
File: '/P4510/sama5d2/poky/meta/lib/oe/sdk.py', lineno: 272, function: _populate
     0268:            self.target_pm.remove_packaging_data()
     0269:
     0270:        bb.note("Installing NATIVESDK packages")
     0271:        self._populate_sysroot(self.host_pm, self.host_manifest)
 *** 0272:        self.install_locales(self.host_pm)
     0273:
     0274:        self.host_pm.run_intercepts(populate_sdk='host')
     0275:
     0276:        execute_pre_post_process(self.d, self.d.getVar("POPULATE_SDK_POST_HOST_COMMAND"))
File: '/P4510/sama5d2/poky/meta/lib/oe/sdk.py', lineno: 104, function: install_locales
     0100:            # Generate a locale archive of them
     0101:            target_arch = self.d.getVar('SDK_ARCH')
     0102:            rootfs = oe.path.join(self.sdk_host_sysroot, self.sdk_native_path)
     0103:            localedir = oe.path.join(rootfs, self.d.getVar("libdir_nativesdk"), "locale")
 *** 0104:            generate_locale_archive(self.d, rootfs, target_arch, localedir)
     0105:            # And now delete the binary locales
     0106:            pkgs = fnmatch.filter(pm.list_installed(), "nativesdk-glibc-binary-localedata-*.utf-8")
     0107:            pm.remove(pkgs)
     0108:        else:
File: '/P4510/sama5d2/poky/meta/lib/oe/package_manager.py', lineno: 135, function: generate_locale_archive
     0131:    # Need to set this so cross-localedef knows where the archive is
     0132:    env = dict(os.environ)
     0133:    env["LOCALEARCHIVE"] = oe.path.join(localedir, "locale-archive")
     0134:
 *** 0135:    for name in sorted(os.listdir(localedir)):
     0136:        path = os.path.join(localedir, name)
     0137:        if os.path.isdir(path):
     0138:            cmd = ["cross-localedef", "--verbose"]
     0139:            cmd += arch_options
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/P4510/sama5d2/poky/build-microchip/tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/sdk/image/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/usr/lib/locale'

ERROR: Logfile of failure stored in: /P4510/sama5d2/poky/build-microchip/tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/temp/log.do_populate_sdk.2746901
ERROR: Task (/P4510/sama5d2/poky/../meta-atmel/recipes-atmel/images/microchip-headless-image.bb:do_populate_sdk) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4956 tasks of which 1680 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /P4510/sama5d2/poky/../meta-atmel/recipes-atmel/images/microchip-headless-image.bb:do_populate_sdk
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

I haven't change any configuration except set `MACHINE ??= "sama5d27-wlsom1-ek-sd"` in `conf\local.conf`.

HiFiPhile, 2021-12-20:
Upon investigation it can be resolved by `GLIBC_SPLIT_LC_PACKAGES = "0"`.
When it's 1 locale's filenames are different than the sdk popolate script:
```
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-hi-in.utf-8-lc-address_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-en-gb.utf-8-lc-ctype_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-de-de.utf-8-lc-identification_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-hi-in.utf-8-lc-time_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-en-gb.utf-8-lc-numeric_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-en-us.utf-8-lc-messages_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-de-de.utf-8-lc-address_2.31-r0_x86_64-nativesdk.ipk
./tmp/work/sama5d27_wlsom1_ek_sd-poky-linux-gnueabi/microchip-headless-image/1.0-r1/oe-sdk-repo/x86_64-nativesdk/nativesdk-glibc-binary-localedata-fr-fr.utf-8-lc-numeric_2.31-r0_x86_64-nativesdk.ipk
```

https://git.yoctoproject.org/poky/tree/meta/lib/oe/sdk.py?h=dunfell#n96

So it looks like a upstream issue. I'm not familiar with Yocto team, could you report the issue ?

OOHehir, 2022-05-02:
Many thanks for posting this solution.

I ran into the same issue with the SDK build for microchip-graphics-image-1.0-2022-03-12:
```
Build Configuration:

BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d4-xplained-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "3.1.15"
TUNE_FEATURES        = "arm vfp cortexa5 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:b9917c90a2001ab4eb09a4dfa301ab8b732f7fa7"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-initramfs       = "dunfell:0722ff6f021df91542b5efa1ff5b5f6269f66add"
meta-atmel           = "dunfell:7080c9bc6c8c898bac7bb1aa0427f3e01394cb3b"
meta-multimedia      = "dunfell:0722ff6f021df91542b5efa1ff5b5f6269f66add"
```

Changing poky/build-microchip/conf/local.conf with your edit resolved the issue.

carolcabral, 2022-11-23:
> Upon investigation it can be resolved by `GLIBC_SPLIT_LC_PACKAGES = "0"`. When it's 1 locale's filenames are different than the sdk popolate script, with a `-lc-*` suffix:

That conf edit worked for `sam9x60ek-sd` and `microchip-graphics-image` as well

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sam9x60ek-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "arm armv5 thumb dsp"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:1cb13d24b592986a44fd653e083b7711513fc916"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-initramfs       = "kirkstone:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-atmel           = "kirkstone:30ef71f26146dfe228c7d4d2c550bc05791ccc0d"
meta-multimedia      = "kirkstone:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-arm             
meta-arm-toolchain   = "kirkstone:67578fcfcd8ee8efcaef67ed7db1dfd55105872e"
```

gehariprasath, 2023-02-02:
Patch addressing this issue is pushed to dunfell &  kirkstone branch. Closing the issue.

closed 2023-02-02 by gehariprasath