linux4sam/meta-atmel #210: plplot forgotten checksum

afinfante, 2021-12-20:
```
afinfante@fenix:~/poky/build$ sync && bitbake microchip-graphics-image
Loading cache: 100% |######################################################################################################################################################################################################################################| Time: 0:00:02
Loaded 3386 entries from dependency cache.
Parsing recipes: 100% |####################################################################################################################################################################################################################################| Time: 0:00:02
Parsing of 2277 .bb files complete (2275 cached, 2 parsed). 3387 targets, 349 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "i686-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d2-xplained"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "3.1.13"
TUNE_FEATURES        = "arm vfp cortexa5 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:795339092f87672e4f68e4d3bc4cfd0e252d1831"
meta-atmel           = "dunfell:20eeec4910f1214b9099f5276a944e5d281b70ee"
meta-oe              
meta-networking      
meta-python          
meta-multimedia      = "dunfell:69f94af4d91215e7d4e225bab54bf3bcfee42f1c"
meta-qt5             = "dunfell:b4d24d70aca75791902df5cd59a4f4a54aa4a125"

Initialising tasks: 100% |#################################################################################################################################################################################################################################| Time: 0:00:21
Sstate summary: Wanted 1590 Found 180 Missed 1410 Current 842 (11% match, 42% complete)
NOTE: Executing Tasks
ERROR: plplot-5.15.0-r0 do_fetch: No checksum specified for '/home/afinfante/poky/build/downloads/PLplot', please add at least one to the recipe:
SRC_URI[sha256sum] = "6007e8dc067a435c79a9994d76efc9e5510a22923fe14e282975e09a7b719257"
ERROR: plplot-5.15.0-r0 do_fetch: Bitbake Fetcher Error: NoChecksumError('Missing SRC_URI checksum', 'https://github.com/PLplot/PLplot;protocol=https')
ERROR: Logfile of failure stored in: /home/afinfante/HMI_BASE_SW/poky/build/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/plp
```

afinfante, 2021-12-20:
This can be solved by adding the following to the plplot recipe
```
SRC_URI[sha256sum] = "6007e8dc067a435c79a9994d76efc9e5510a22923fe14e282975e09a7b719257"
```

afinfante, 2021-12-20:
There are also errors with the management of plplot licenses in the recipe, I am opening a pull request to correct it.

afinfante, 2021-12-20:
See X for more information

Dharma-B, 2022-12-01:
This Issue is not reproducible with the latest kirkstone branch, hence closing it.

closed 2022-12-01 by Dharma-B