linux4sam/meta-atmel #212: egt-benchmark: git submodules are not initialized

priv-kweihmann, 2021-12-24:
tool is using an external dependency https://github.com/linux4sam/egt-benchmark/tree/master/external, which doesn't seem to be fully initialized when using the git-fetcher
compilation fails with `src/main.cpp:7:10: fatal error: 'hayai.hpp' file not found`

ldesroches, 2022-01-05:
Hello,

How can I reproduce your issue, I have run a build and I have no issue to build this package.

Regards,

priv-kweihmann, 2022-01-05:
Requires a not yet patch in bitbake (see https://lists.openembedded.org/g/openembedded-core/message/159995, resp. https://errors.yoctoproject.org/Errors/Build/137579/) - it isn't a problem as of now, but likely will break once upstream merges that feature. IMO that should happen very soon.

In my opinion the recipe should use the gitsm fetcher instead of plain git

Dharma-B, 2022-12-01:
Fixes: https://github.com/linux4sam/meta-atmel/commit/492c4937c72fe53971f409746b6cdcb3f3dd754c

closed 2022-12-01 by Dharma-B