linux4sam/meta-atmel #213: Custom bootstrap V4.x builds

elockman, 2022-01-03:
In the last month, the at91bootstrap has moved from v3x to v4x.  This brings a device-centric configuration rather than a board-centric configuration.  It reduces a lot of duplicate code, but I am having an issue adding a bootstrap v4.x config for a custom board.
Can you point me to a document that explains adding a bootstrap v4.x board configuration for yocto builds?

I am using a custom sama5d2 board which is a bit of a hybrid between the xplained board and the SOM-EK board.

"newboard" is a generic term here, but this is what I've tried, by forking the boostrap project and pointing to my edited repo.

I've added newboard to the COMPATIBLE_MACHINE definition.

Next, I've added a newboard_qspi_uboot_defconfig file to the at91bootstrap/configs folder.

Finally, I've added to these definitions in the at91bootstrap.inc file
AT91BOOTSTRAP_CONFIG_newboard ??= "${AT91BOOTSTRAP_MACHINE}_qspi_uboot"
AT91BOOTSTRAP_LOAD_newboard ??= "qspiboot-uboot"

My build is failing with the following error:
ERROR: at91bootstrap-4.0.1+gitAUTOINC+905ba56436-r0 do_configure: No config files found

What am I missing?

mattwood-microchip, 2022-01-03:
It sounds like your trying to directly modify the meta-atmel layer with your board definitions.  The defconfig you added to your at91bootstrap configs folder would only be local and not upstream so that could be part of the problem if you did a bitbake -c clean at91bootstrap or something similar and sources were downloaded and extracted.  

Why not create a custom meta layer for you project instead and keep your board specific modifications outside of the meta-atmel layer?  You would have the option to create your own machine definition to handle things specific to your hardware design for example external NVM arguments like ubi for NAND flash if you wanted.  It would make things easier in my opinion.

As an example, if your custom layer is called meta-newboard:

meta-newboard
├── conf
│   ├── layer.conf
│   └── machine
│       └── newboard.conf
├── COPYING.MIT
├── README
├── recipes-bsp
│   ├── at91bootstrap
│   │   ├── at91bootstrap_4.0.1.bbappend
│   │   └── files
│   │       └── defconfig
│   └── u-boot
│       ├── files
│       │   ├── 0001-Add-initial-newboard-support.patch
│       │   ├── defconfig
│       │   └── envs
│       │       └── newboard.txt
│       └── u-boot-at91_2021.04.bbappend
└── recipes-kernel
    └── linux
        ├── linux-at91-5.10.50
        │   ├── 0001-Add-initial-newboard-support.patch
        │   ├── 0002-Add-NAND-support-for-newboard.patch
        │   └── newboard
        │       └── defconfig
        └── linux-at91_5.10.50.bbappend

The bbappend for bootstrap and others would look something like this:
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

COMPATIBLE_MACHINE_dorado = "(dorado)"

SRC_URI +=	" \
			file://defconfig \
		"

elockman, 2022-01-03:
Thank you for your quick response. I do have a custom layer like meta-newboard.  I have been using this for over a year.  Something was broken when the bootstrap was updated, and I've had trouble resolving it. I've temporarily resolved the issue by reverting the bootstrap to v3x, but I would like to move forward with v4x.

My question really relates to configuring the bootstrap v4x.  Am I missing any files which need to changed for my new configuration?

I understand the  FILESEXTRAPATHS_prepend := "${THISDIR}/files:" change

And, I can add SRC_URI += newboard__qspi_uboot_defconfig to the files folder

I can add COMPATIBLE_MACHINE_newboard = "(newboard)" to the at91bootstrap_4.0.1.bbappend file

In the custom meta-newboard context, bootstrap uses an .inc file.  What is the best way to edit the bootstrap .inc file to include these definitions?
AT91BOOTSTRAP_CONFIG_newboard ??= "${AT91BOOTSTRAP_MACHINE}_qspi_uboot"
AT91BOOTSTRAP_LOAD_newboard ??= "qspiboot-uboot"

I hope my questions make sense.  Thanks, again.