linux4sam/meta-atmel #214: tslib inclusion with libegt

OOHehir, 2022-01-23:
Hello,

Following issues compiling tslib support into libegt it appears IMO there are one/ maybe two issues with the current bitbake [recipe ](https://github.com/linux4sam/meta-atmel/blob/20eeec4910f1214b9099f5276a944e5d281b70ee/recipes-graphics/libegt/libegt_1.3.bb)(for libegt_1.X.bb)

1. As tslib is not included in DEPENDS it seems its presence will never be discovered. 
2. Unless ```EXTRA_OECONF_append +="--with-tslib"``` is included in a bbappend the current config will test for the presence of tslib & append a --without-tslib to the output CONF which will hence disregard any use of ```EXTRA_OECONF +="--with-tslib"```

I'm sure there's a cleaner way of testing for the presence tslib & compiling it into libegt but my bbappend (which compiles in tslib) is:
```
DEPENDS_append +="tslib"
EXTRA_OECONF_append +="--with-tslib"
```

codrin989, 2022-01-26:
Hi @OOHehir . Please try including tslib in libegt's PACKAGECONFIG.

OOHehir, 2022-01-26:
Thanks for the response @codrin989. In libegt's bitbake recipe there's currently a line

```PACKAGECONFIG[tslib] = "--with-tslib,--without-tslib,tslib"```

Do you mean to replace it with:

```PACKAGECONFIG[tslib] = "--with-tslib"```

codrin989, 2022-01-26:
No, I mean https://github.com/linux4sam/meta-atmel/blob/dunfell/recipes-graphics/libegt/libegt_1.3.bb#L31 . Please see https://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#var-PACKAGECONFIG on how to modify libegt's PACKAGECONFIG on your side.

OOHehir, 2022-01-26:
Ok, many thanks. 

For others looks like this works in a bbappend file:

```
PACKAGECONFIG_append = " tslib"
```

closed 2022-01-26 by OOHehir