linux4sam/meta-atmel #218: Non GUI Image for SAM9X60

ankitsingh1508, 2022-03-12:
Hi,

Tried "microchip-graphics-image" for SAM9X60 and it boots and works fine with GUI display but when tried other images like "core-image-minimal" then nothing comes on screen (i was expecting linux terminal login screen).
Am i missing some thing to be added to local.conf for this...?

Regards.

mattwood-microchip, 2022-03-12:
Core-image-minimal is a console image target with stdio on a USART connected to the virtual com port on the onboard jlink debugger (J22).  If you really want to redirect to the display you need to enable the fbcon driver and update the kernel command line boot args.  You can Google more about it or reference the kernel docs below.
https://docs.kernel.org/fb/fbcon.html

ankitsingh1508, 2022-03-22:
@mattwood-microchip 
Thanks a lot for the guidance.

closed 2023-04-13 by noglitch