linux4sam/meta-atmel #223: Mesh not supported in linux-mchp_5.10.80?

elockman, 2022-05-17:
I am building a linux image with yocto for the SAMA5D2-XULT board.
I am trying to assign mesh type to an RT5370 USB wifi device using the rt2800usb driver.
This has worked in the past, but mesh is no longer supported for some reason.

root@sama5d2-xplained:~# iw wlan0 interface add mesh0 type mp mesh_id mymesh
**command failed: Operation not supported (-95)**

iw list yields:

        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor

mesh point interface is absent.
I suspect my mesh config is not being applied properly with the new Linux-mchp version.

I am building with linux-mchp_5.10.80

my linux-mchp_5.10.80.bbappend file includes:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SRC_URI_append = " file://xult_dts.patch;patch=1"
SRC_URI_append += " file://defconfig.patch"
```

and the defconfig patch includes:
```
diff --git a/arch/arm/configs/at91_dt_defconfig b/arch/arm/configs/at91_dt_defconfig
--- a/arch/arm/configs/at91_dt_defconfig
+++ b/arch/arm/configs/at91_dt_defconfig
@@ -263,3 +263,6 @@
 # CONFIG_SCHED_DEBUG is not set
 # CONFIG_FTRACE is not set
 CONFIG_DEBUG_USER=y
+# MESH CHANGES
+CONFIG_MAC80211_MESH=y
+CONFIG_BRIDGE=y
```

The image builds, but the mesh configuration does not seem to stick.
Removing and re-inserting the module, shows that Linux is recognizing the module

root@sama5d2-xplained:~# usb 1-2: USB disconnect, device number 2
usb 1-2: new high-speed USB device number 3 using atmel-ehci
usb 1-2: New USB device found, idVendor=148f, idProduct=5370, bcdDevice= 1.01
usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-2: Product: 802.11 n WLAN
usb 1-2: Manufacturer: Ralink
usb 1-2: SerialNumber: 1.0
usb 1-2: reset high-speed USB device number 3 using atmel-ehci
ieee80211 phy4: rt2x00_set_rt: Info - RT chipset 5390, rev 0502 detected
ieee80211 phy4: rt2x00_set_rf: Info - RF chipset 5370 detected
ieee80211 phy4: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
ieee80211 phy4: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36

I've also removed the driver with rmmod and reinstalled with modprobe
rmmod rt2800usb
modprobe rt2800usb nohwcrypt=0

lsmod shows the driver is installed.

```
root@sama5d2-xplained:~# lsmod
Module                  Size  Used by
rt2800usb              24576  0
8021q                  20480  0
usb_f_acm              12288  1
u_serial               16384  1 usb_f_acm
g_serial               12288  0
libcomposite           32768  2 g_serial,usb_f_acm
rt2800lib             102400  1 rt2800usb
crc_ccitt              12288  1 rt2800lib
rt2x00usb              16384  1 rt2800usb
rt2x00lib              40960  3 rt2800lib,rt2800usb,rt2x00usb
mac80211              368640  3 rt2800lib,rt2x00lib,rt2x00usb
cfg80211              278528  2 rt2x00lib,mac80211
atmel_usba_udc         20480  0

```
Does mesh need to be configured differently in linux-mchp_5.10.80?

elockman, 2022-05-17:
If not clear, I am using the default dunfell branch.

I've added the following to to the end of my local.conf, but yocto is still building 5.10.80.
```
PREFERRED_PROVIDER_virtual/kernel = "linux-at91"
PREFERRED_VERSION_linux-at91 = "5.4.81%"
```
@noglitch, how should I force linux v5.4.81 in the dunfell branch?

alexandrebelloni, 2022-05-17:
You need to use

`PREFERRED_PROVIDER_virtual/kernel_sama5 = "linux-at91"`

elockman, 2022-05-18:
By forcing linux version 5.4.81, mesh point returns.
```
root@sama5d2-xplained:~# uname -a
Linux sama5d2-xplained 5.4.81-linux4sam-2020.10 #1 Thu Jan 14 12:54:56 UTC 2021 armv7l armv7l armv7l GNU/Linux
```
```
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
```

noglitch, 2022-05-18:
Even if we merged one additional driver changes in the network/WiFi sub-system (wilc1000/3000), I do think that we didn't touch this rt2800usb driver. I suspect it's a change in the upstream kernel.
So I'm closing this issue.
Best regards,
  Nicolas

closed 2022-05-18 by noglitch