linux4sam/meta-atmel #225: Using KBUILD_DEFCONFIG not straightforward

rob-baily, 2022-08-19:
I have been trying to use a custom defconfig in the Linux kernel build for Yocto using KBUILD_DEFCONFIG in a bbappend file and have been struggling with it.    I believe I finally got to the root of the issue.  In the recipes-kernel [linux.inc](https://github.com/linux4sam/meta-atmel/blob/dunfell/recipes-kernel/linux/linux.inc) there is this code:

```
do_configure_prepend() {
	if [ ! -f "${WORKDIR}/defconfig" ] && [ -n "${KBUILD_DEFCONFIG}" ]; then
		if [ -f "${S}/arch/${ARCH}/configs/${KBUILD_DEFCONFIG}" ]; then  
			cp -f ${S}/arch/${ARCH}/configs/${KBUILD_DEFCONFIG} ${WORKDIR}/defconfig
                else
                        bbfatal "A KBUILD_DEFCONFIG '${KBUILD_DEFCONFIG}' was specified, but not present in the source tree"
                fi
        fi
}
```

There are 2 issues that I had:

- The recipe linux-mchp_5.15.32.bb includes the line `SRC_URI_append_sama5 += "file://defconfig"` so the first part of the if condition is always true because defconfig is included in SRC_URI so the KBUILD_DEFCONFIG value was not being used.
- If the custom defconfig was to change it would not be modified because of the first condition checking if the file already exists.

I had to work around this issue by including the line `SRC_URI_remove_sama5 += "file://defconfig"` in my bbappend file to be able to to use KBUILD_DEFCONFIG but this was not obvious based on what I saw in the recipe and include file.   I still think this does not address the second bullet where the file already exists.

I am thinking that it would be best to remove the first check on whether the defconfig exists so that if KBUILD_DEFCONFIG it is always processed.  Thoughts or questions?  If it seems good I can submit a pull request.  Another option might be to put a KBUILD_DEFCONFIG file straight to ,config because in the core poky meta kernel.bbclass there is a note in the kernel_do_configure function: 

```
# Copy defconfig to .config if .config does not exist. This allows
# recipes to manage the .config themselves in do_configure_prepend().

rob-baily, 2022-09-06:
@noglitch I was wondering if you had a chance to review this?  I would be happy to submit a pull request.  I am thinking now that the second option I put forth to copy the KBUILD_DEFCONFIG defined file to .config would match up better with the kernel class recipe.

Dharma-B, 2022-11-30:
Hi @rob-baily,
It was intentionally set that In case KBUILD_DEFCONFIG is set and defconfig file is present, the defconfig file takes priority and KBUILD_DEFCONFIG is ignored, because the defconfig file in meta-atmel may contain some necessary configs for the specified board that are not upstreamed yet.

If you want to use a custom defconfig you can simply include the line SRC_URI:append:sama5 = " file://defconfig" in your bbappend file by introducing a custom defconfig in your layer which would probably replace the defconfig from meta-atmel layer.

@gehariprasath your thoughts?

alexandrebelloni, 2022-11-30:
I've submitted https://github.com/linux4sam/meta-atmel/pull/229 which removes KBUILD_DEFCONFIG but keeps the same functionality. I'd love SoC vendors to stop rolling out their custom do_config...

Dharma-B, 2023-02-06:
Thanks for the PR @alexandrebelloni, It has been merged, hence closing this issue.