linux4sam/meta-atmel #228: SAMA5D2 XULT missing nat and mangle tables in iptables

elockman, 2022-11-15:
I am not able to forward comms through a device using iptables. The setup looks like the following:

INTERNET-----ROUTER---------------------------RED----------------------------------WHITE
.........................(eth0).............................(eth0)...(wlan0).......................(wlan0)
......................192.168.1.1........192.168.1.138.....192.168.4.51............192.168.4.52

Red should be able to route comms between the wlan0 (192.168.4.0) subnet and the eth0 (192.168.1.0) subnet.

Red can ping White wlan0 (192.168.4.52), the router (192.168.1.1) and google (8.8.8.8)
White can ping Red wlan0 (192.168.4.51), Red eth0 (192.168.1.138), but **NOT** the router (192.168.1.1) and **NOT** google (8.8.8.8) 

I am working with the dunfell branch:
```
root@red:~# uname -a
Linux red 5.4.81-linux4sam-2020.10 #1 Thu Jan 14 12:54:56 UTC 2021 armv7l armv7l armv7l GNU/Linux
```
Red has ip_forward set, verified with:
```
root@red:~# cat /proc/sys/net/ipv4/ip_forward
1
```
iptables version:
```
root@red:~# iptables -V
iptables v1.8.4 (legacy)
```
The default filter table is empty, but has the ACCEPT policy:
```
root@red:~# iptables -t filter -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
```
The nat, mangle, and raw tables are not found:
```
root@red:~# iptables -t nat -L
iptables v1.8.4 (legacy): can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@red:~# iptables -t mangle -L
iptables v1.8.4 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@red:~# iptables -t raw -L
iptables v1.8.4 (legacy): can't initialize iptables table `raw': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
```
nftables is no help either.
```
root@red:~# nft
../../nftables-0.9.6/src/mnl.c:45: Unable to initialize Netlink socket: Protocol not supported
```
**In order to route packets, I will need the nat table. That seems to be a kernel config in yocto.** 

Is iptables or nftables recommended for the sama5d2 xplained board? How do I add the nat table to iptables or how is the nftables built properly?

elockman, 2022-11-16:
Looking at https://github.com/linux4sam/linux-at91/tree/master/kernel/configs,
These appear to be what I'm looking for. Are these compatible with SAMA5D2?
```
CONFIG_IP_ADVANCED_ROUTER=y
CONFIG_IP_MULTICAST=y
CONFIG_IP_MULTIPLE_TABLES=y
CONFIG_IP_NF_ARPFILTER=y
CONFIG_IP_NF_ARPTABLES=y
CONFIG_IP_NF_ARP_MANGLE=y
CONFIG_IP_NF_FILTER=y
CONFIG_IP_NF_IPTABLES=y
CONFIG_IP_NF_MANGLE=y
CONFIG_IP_NF_MATCH_AH=y
CONFIG_IP_NF_MATCH_ECN=y
CONFIG_IP_NF_MATCH_TTL=y
CONFIG_IP_NF_NAT=y
CONFIG_IP_NF_RAW=y
CONFIG_IP_NF_SECURITY=y
CONFIG_IP_NF_TARGET_MASQUERADE=y
CONFIG_IP_NF_TARGET_NETMAP=y
CONFIG_IP_NF_TARGET_REDIRECT=y
CONFIG_IP_NF_TARGET_REJECT=y
```

noglitch, 2022-11-16:
Hi,
Yes, these are the config options that you need to add to your kernel defconfig.
They also can be compiled as modules, so you even can insert them in your already built system at the condition that you use exactly the same source code as the binary used.
Best regards,
  Nicolas

elockman, 2022-11-17:
If anyone in the future is curious, I went through the code and added all CONFIG's I could find. Likely, only a few are needed, but I just wanted to make sure my bases were covered. Mine is application is now working.

Code reviewed here for configs:
https://github.com/torvalds/linux/blob/master/net/ipv4/netfilter/Kconfig

Added configs:
```
CONFIG_INET=y
CONFIG_NETFILTER=y
CONFIG_NETFILTER_ADVANCED=y
CONFIG_NF_CONNTRACK=y
CONFIG_NF_CONNTRACK_SNMP=y

CONFIG_NF_DEFRAG_IPV4=y
CONFIG_NF_SOCKET_IPV4=y
CONFIG_NF_TPROXY_IPV4=y

CONFIG_NF_NAT=y
CONFIG_NF_NAT_SNMP_BASIC=y
CONFIG_NF_NAT_PPTP=y
CONFIG_NF_NAT_H323=y
CONFIG_IP_NF_IPTABLES=y
CONFIG_IP_NF_MATCH_AH=y
CONFIG_IP_NF_MATCH_ECN=y
CONFIG_IP_NF_MATCH_RPFILTER=y
CONFIG_IP_NF_MATCH_TTL=y
CONFIG_IP_NF_FILTER=y
CONFIG_IP_NF_TARGET_REJECT=y
CONFIG_IP_NF_TARGET_SYNPROXY=y
CONFIG_IP_NF_NAT=y
CONFIG_IP_NF_TARGET_MASQUERADE=y
CONFIG_IP_NF_TARGET_NETMAP=y
CONFIG_IP_NF_TARGET_REDIRECT=y
CONFIG_IP_NF_MANGLE=y
CONFIG_IP_NF_TARGET_CLUSTERIP=y
CONFIG_IP_NF_TARGET_ECN=y
CONFIG_IP_NF_TARGET_TTL=y
CONFIG_IP_NF_RAW=y
CONFIG_IP_NF_SECURITY=y
CONFIG_IP_NF_ARPTABLES=y
CONFIG_IP_NF_ARPFILTER=y
CONFIG_IP_NF_ARP_MANGLE=y 
```

closed 2022-11-17 by elockman