linux4sam/meta-atmel #232: linux4sam.org build instructions need to be revised

elockman, 2022-12-08:
Hello,

The instructions for **sama5d2Xplained** are showing build instructions for the **sam9x60-curiosity-sd** (This might be true for all boards)
https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d2XplainedMainPage

```
Typical bitbake output
======================
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sam9x60-curiosity-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "arm armv5 thumb dsp"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-yocto-bsp       = "heads/kirkstone-4.0.5:2e79b199114b25d81bfaa029ccfb17676946d20d"
meta-oe              
meta-networking      
meta-webserver
meta-python          
meta-initramfs       = "kirkstone:744a4b6eda88b9a9ca1cf0df6e18be384d9054e3"
meta-atmel           = "kirkstone:e71f4b71dba523c464cbb7aaa374b3e5e92bb674"
meta-multimedia      = "kirkstone:744a4b6eda88b9a9ca1cf0df6e18be384d9054e3"
meta-arm
meta-arm-toolchain   = "kirkstone:bafd1d013c2470bcec123ba4eb8232ab879b2660"

```
If you take a ride on the way-back-machine, this is what you see from the dunfell branch:
https://web.archive.org/web/20220811055758/https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d2XplainedMainPage
```
Typical bitbake output
======================
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d27-wlsom1-ek-sd"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "3.1.15"
TUNE_FEATURES        = "arm vfp cortexa5 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:b41d4e46d30ed32e11d68466721a0f4056bad700"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-initramfs       = "dunfell:95969f0f5f4264644a5dbfb6d4fb66e2ac3a2cde"
meta-atmel           = "dunfell:149de95130a392fd7868cfab555ee377b5ab6459"
meta-multimedia      = "dunfell:95969f0f5f4264644a5dbfb6d4fb66e2ac3a2cde"
```
Some of the github clone URLs have changed from **https:** to **git:** also. This may be noteworthy, because github now requires clones from https instead of http urls.

Also odd is the revision note at the bottom of the page shows 2018:
`r8 - 23 Aug 2018 - 13:06:31 - [RazvanStefanescu]`

@noglitch @ehristev @RazvanStef can these instructions be updated for specific boards, or at least similar chipsets?

noglitch, 2022-12-08:
Hi @elockman,
Thanks for the feedback as it contains lot of information for us to address.
Note that, before going into details, we heavily use an "include" mechanism to build Linux4SAM board pages, and that leads to having "generic" instructions for all the boards. Typically, Yocto Project build procedure is the same for all our boards and basically that's why we mention "**Typical** bitbake output", which is not the "**exact** bitbake output for the given board".
So now going further:

1. we'll keep "typical" bitbake output and Yocto Project instruction "example" for one board that could be adapted for all of the MACHINEs listed at the top of the procedure (which is basically the README file for the Yocto Project layer)
2. some of the instructions must be updated and your highlight of git: vs. https: is a very valid point: we'll fix that
3. the fact that you see "old" revision notes at the bottom of pages is precisely the consequence of this use of "include" mechanism in our wiki instance: the "template" page is kept unchanged whereas the instructions are updated (see [PokyBuild?raw=y](https://www.linux4sam.org/bin/view/Linux4SAM/PokyBuild?raw=y) %STARTSECTION{"build_poky"}% with a revision which is very new)

Hope that it helps clarifying. Don't hesitate to come back to us for more remarks.
Best regards,
  Nicolas

elockman, 2022-12-08:
Thank you @noglitch.  

I'm upgrading from dunfell to kirkstone.
I've had a bit of trouble with it.  Some of it is due to the syntax changes, some are still unknown.  I'm trying to go back to the minimal build as a sanity check.

Thanks again for the response and for taking a second look.

closed 2023-03-01 by elockman