linux4sam/meta-atmel #237: Kirkstone pixman error in .bbappend

grahas, 2023-01-23:
In meta-atmel/recipes-graphics/pixman/pixman_%.bbappend since it is appending it should have a space in the beginning or it will cause issues with that variable. ie SRC_URI:append = "file://0001-Revert-Initialize-temporary-buffers-in-general_compo.patch" -> SRC_URI:append = " file://0001-Revert-Initialize-temporary-buffers-in-general_compo.patch"

https://docs.yoctoproject.org/bitbake/dev/bitbake-user-manual/bitbake-user-manual-metadata.html#appending-and-prepending-override-style-syntax

gehariprasath, 2023-02-01:
@grahas While it didn't throw any error/warnings when I tried to build pixman, I agree with you that this syntax not correct and a space needs to precede the file name. You can send me a PR if you would like to or I can fix this myself with your name in the Reported-by tag. Let me know how you would want to proceed. Thanks for noticing this.

grahas, 2023-02-02:
Hi @gehariprasath, I made PR Update Append Syntax #238 . Thanks for helping take care of this :+1: