linux4sam/meta-atmel #239: master is not maintained

alexandrebelloni, 2023-02-06:
The master branch is not maintained and will not work properly, it should be removed from the repository.

noglitch, 2023-02-07:
@alexandrebelloni : but still, we are trying... Our latest updates are from beginning of January and yes, surely not complete to reach a working condition.
Anyway, I would like to benefit from the fact of this master branch being tried and tested, so I would prefer not to remove it.

alexandrebelloni, 2023-02-07:
What is being tested is not working then because you can very easily see that the layer has not been converted to the new override syntax properly, whereas the kirkstone branch is.

gehariprasath, 2023-02-09:
@alexandrebelloni The master branch is now synced with kirkstone and tested with our boards. So I am closing this now. Pls feel free to open a new issue if you see anything abnormal. Thank you.

closed 2023-02-09 by gehariprasath