linux4sam/meta-atmel #244: Using USB wireless modules with Kirkstone branch

elockman, 2023-02-27:
I am trying to use wireless USB modules with the new kirkstone yocto branch on the at91sama5d27-XULT board, but I am not having any luck.

I am using Ralink 5370 (which worked in dunfell)
I am also using MT7601 and R8188 as candidate successors to the legacy Ralink5370 USB WiFi.

I've used the following video as a loose guide:
https://www.youtube.com/watch?v=OlMY7iDoN_w

`ifconfig -a` does not yield any wlan0 interfaces. (only eth0, lo, and sit0)

I used menuconfig to edit my defconfig file per the video above.
The mac80211 and cfg80211 modules did not load originally, but I have them loaded now.

lsusb yields:
```
root@sama5d2-xplained:/# lsusb
Bus 001 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 002: ID 0bda:f179 Realtek Semiconductor Corp. RTL8188FTV 802.11b/g/n 1T1R 2.4G WLAN Adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
lsmod yields:
```
root@sama5d2-xplained:~# lsmod
Module                  Size  Used by
g_serial               12288  0
libcomposite           36864  1 g_serial
mac80211              380928  0
cfg80211              286720  1 mac80211
```
I am not seeing *.ko files for the usb drivers. I am only seeing Kconfig and MakeFile files for them.
```
root@sama5d2-xplained:/lib/modules/5.15.68-linux4microchip-2022.10/build/drivers/staging/r8188eu# ls
Kconfig  Makefile
root@sama5d2-xplained:/lib/modules/5.15.68-linux4microchip-2022.10/build/drivers/net/wireless/mediatek/mt7601u# ls
Kconfig  Makefile
root@sama5d2-xplained:/lib/modules/5.15.68-linux4microchip-2022.10/build/drivers/net/wireless/ralink/rt2x00# ls
Kconfig  Makefile
```
What must I do in order to build these drivers? How can I debug this further?

elockman, 2023-03-01:
My new defconfig file includes:
```
CONFIG_RT2X00=m
CONFIG_RT2500USB=m
CONFIG_RT73USB=m
CONFIG_RT2800USB=m
CONFIG_RT2800USB_RT53XX=y
CONFIG_RT2800USB_RT55XX=y
CONFIG_RT2800USB_UNKNOWN=y
CONFIG_MT7601U=m
CONFIG_R8188EU=m
CONFIG_88EU_AP_MODE=y
```
defconfig attached as a text file:
[defconfig.txt](https://github.com/linux4sam/meta-atmel/files/10862682/defconfig.txt)

What commands do I need to include the firmware and kernel modules?
I assume I need something like the following:
```
IMAGE_INSTALL += "kernel-module-mt7601u"
IMAGE_INSTALL += "firmware-mt7601u"
IMAGE_INSTALL += "kernel-module-r8188eu"
IMAGE_INSTALL += "linux-firmware-rtl8188"
```

elockman, 2023-03-01:
This shows all 3 of my devices responding to lsusb:
```
root@sama5d2-xplained:/# lsusb
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 002: ID 0bda:f179 Realtek Semiconductor Corp. RTL8188FTV 802.11b/g/n 1T1R 2.4G WLAN Adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

elockman, 2023-03-02:
The kernel modules are yielding errors:
```
 * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'kernel-module-r8188eu'.
 AND
 *   - nothing provides kernel-module-mt7601u needed by packagegroup-machine-base-1.0-r83.sama5d2_xplained
```
@Dharma-B do you have any suggestions? What might I be missing, or what may be needed for these kernel modules to be built?

Dharma-B, 2023-03-02:
Hi @elockman 
Please use the following patch and let me know if the issue still persists.
[0001-Using-USB-wireless-modules-with-Kirkstone-branch.txt](https://github.com/linux4sam/meta-atmel/files/10870715/0001-Using-USB-wireless-modules-with-Kirkstone-branch.txt)

elockman, 2023-03-02:
I've been running  **bitbake core-image-base**

**bitbake microchip-headless-image** is giving me the following error:
```
ERROR: qemu-native-6.2.0-r0 do_compile: oe_runmake failed
ERROR: qemu-native-6.2.0-r0 do_compile: ExecutionError('/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/temp/run.do_compile.1150293', 1, None, None)
ERROR: Logfile of failure stored in: /home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/temp/log.do_compile.1150293
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 48 LD=ld  AR=ar OBJCOPY=objcopy LDFLAGS=-L/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/usr/lib                         -L/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/circleci/project/yocto/poky/build-microchip/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -fuse-ld=bfd
| /home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/usr/bin/ninja  build.ninja && touch build.ninja.stamp
| ninja: no work to do.
| /home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/qemu-native/6.2.0-r0/recipe-sysroot-native/usr/bin/meson introspect --targets --tests --benchmarks | /home/circleci/project/yocto/poky/build-microchip/tmp/hosttools/python3 -B scripts/mtest2make.py > Makefile.mtest
```

**bitbake atmel-demo-image** is giving me the following error:
```
ERROR: gcc-11.3.0-r0 do_compile: oe_runmake failed
ERROR: gcc-11.3.0-r0 do_compile: ExecutionError('/home/circleci/project/yocto/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/gcc/11.3.0-r0/temp/run.do_compile.2388256', 1, None, None)
ERROR: Logfile of failure stored in: /home/circleci/project/yocto/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/gcc/11.3.0-r0/temp/log.do_compile.2388256
```

and **bitbake core-image-base** is giving me the following error:
```
ERROR: core-image-base-1.0-r0 do_rootfs: Unable to install packages. Command '/home/circleci/project/yocto/poky/build-microchip/tmp/work/sama5d2_xplained-poky-linux-gnueabi/core-image-base/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /home/circleci/project/yocto/poky/build-microchip/tmp/work/sama5d2_xplained-poky-linux-gnueabi/core-image-base/1.0-r0/opkg.conf -t /home/circleci/project/yocto/poky/build-microchip/tmp/work/sama5d2_xplained-poky-linux-gnueabi/core-image-base/1.0-r0/temp/ipktemp/ -o /home/circleci/project/yocto/poky/build-microchip/tmp/work/sama5d2_xplained-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install apt bridge-utils curl czmq dhcpcd dnsmasq dpkg git gnupg gnutls gssdp hostapd inotify-tools iperf3 iw iwd keepalived kernel-devicetree kernel-devsrc kernel-image kernel-module-mt7601u kernel-module-r8188eu kernel-modules libmicrohttpd libmodbus libnss-mdns linux-firmware-mt7601u linux-firmware-rtl8188 lshw mbedtls mdns ntp openssh openssl packagegroup-base-extended packagegroup-core-boot packagegroup-core-buildessential psplash python3 python3-dev python3-periphery python3-pip python3-psutil quagga run-postinsts sqlite3 usbutils wireless-regdb-static wpa-supplicant zlib' returned 255:
 * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'kernel-module-mt7601u'.
```

Does it make sense to add some packages to core-image-base to allow bitbake to build? I'm not sure which packages those would be.

elockman, 2023-03-03:
If I remove the patches and **bitbake microchip-headless-image**, I still get this error:
```
ERROR: rust-llvm-native-1.59.0-r0 do_compile: ExecutionError('/home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/rust-llvm-native/1.59.0-r0/temp/run.do_compile.1123625', 1, None, None)
ERROR: Logfile of failure stored in: /home/circleci/project/yocto/poky/build-microchip/tmp/work/x86_64-linux/rust-llvm-native/1.59.0-r0/temp/log.do_compile.1123625
```
Is this a separate existing upstream issue? I see this posted elsewhere online for other linux builds?

@Dharma-B, what other useful info can I provide about my system?

Dharma-B, 2023-03-03:
Hi @elockman 
Can you please delete all other directories in build-microchip except downloads/ and double check the branches of other layers (poky,oe,..) once and trigger a build for headless image ?

We haven't faced the above issues, probably some procedural mistake.

elockman, 2023-03-04:
I've tried to strip everything back to the original code with no patches, but I'm essentially getting the same result.
This is my bitbake output:
```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "sama5d2-xplained"
DISTRO               = "poky-atmel"
DISTRO_VERSION       = "4.0.8"
TUNE_FEATURES        = "arm vfp cortexa5 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:a361fb3df9c87cf12963a9d785a9f99faa839222"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-initramfs       = "kirkstone:82c75b466e55d7dca7a2364986ecb704cf63d141"
meta-atmel           = "kirkstone:772408b60c483e6157077699c42f10e1b817029d"
meta-multimedia      = "kirkstone:82c75b466e55d7dca7a2364986ecb704cf63d141"
meta-arm             
meta-arm-toolchain   = "kirkstone:a0216a41bde3f191037332a9238648bedfbc61bd"
```
I am following the instructions from the [linux4sam](https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d2XplainedMainPage#Build_Yocto_Poky_rootfs_from_sou) website.

It is a little unclear whether the documented repos are correct, however. The documentation has a branch discrepancy for the meta-atmel branch. It says **master** branch in one place and **kirkstone** branch in another. This is what I am using:
```
        git clone git://git.yoctoproject.org/poky -b kirkstone yocto/poky
        git clone git://git.openembedded.org/meta-openembedded -b kirkstone yocto/meta-openembedded
        git clone https://github.com/aws/meta-aws.git -b kirkstone yocto/meta-aws
        git clone https://github.com/linux4sam/meta-atmel.git -b kirkstone yocto/meta-atmel
        git clone git://git.yoctoproject.org/meta-arm -b kirkstone yocto/meta-arm
```
I am also using the sample local.conf file from [here](https://github.com/linux4sam/meta-atmel/blob/kirkstone/conf/local.conf.sample).

This works:
> bitbake core-image-base

These do not:
> bitbake atmel-demo-image
> bitbake microchip-headless-image

@Dharma-B, can you verify that the bitbake output looks correct and that the correct repos are being cloned?

Dharma-B, 2023-03-06:
@elockman Thanks for pointing out the flaws in the document, I have updated the page with the latest README from the meta-atmel repo.

I did a clean build to reproduce the issue which you are facing but no luck, the build succeeded for me.

Could you please check if you have installed all the essential build tools and share the full error log so that I can have a look?

elockman, 2023-03-10:
@Dharma-B, I am still having issues with the build. I am doing clean builds online with circleci.com with a ubuntu 20.04 docker image. The image is a full clean build. I've attached the docker file I am using along with the build output. 
[dockerfile.txt](https://github.com/linux4sam/meta-atmel/files/10937699/dockerfile.txt)
[buildOutput.txt](https://github.com/linux4sam/meta-atmel/files/10937700/buildOutput.txt)

I've had to add the following package dependencies:
`apt-get install -qy locales gcc-arm-linux-gnueabi zstd liblz4-tool xxd file git openssh-client`
, but I still fail with the rust-llvm-native error. Using ubuntu 22.04, I find that the pylint3 package is not found. Additionally, I find I can build the **atmel-demo-image** by removing the following:
```
    gdb \
    kea \
    libdrm \
    bluez5 \
    tcpdump \
```
The following lines also yield errors individually:
```
IMAGE_INSTALL += "kernel-module-mt7601u"
IMAGE_INSTALL += "kernel-module-r8188eu"
```
If you look above, I've been including the https://github.com/aws/meta-aws.git repo. I have since removed that from my build.

Please tell me which distribution you are building on (i.e. ubuntu 18.04?).

elockman, 2023-03-10:
@Dharma-B I found a similar issue posted online with an NXP yocto build here:
https://community.nxp.com/t5/i-MX-Processors/IMX8MM-in-Yocto-5-15-52-rust-llvm-1-59-0-bb-do-compile-failed/m-p/1568799

The proposed fix is to update to rust-llvm 1.67.1, but kirkstone is meant to be fixed to rust-llvm 1.59.0 as seen here:
https://layers.openembedded.org/layerindex/recipe/300320/

If you provide your pertinent build environment details (OS, dependencies, and repos), I would be happy to try to recreate the standard build.  Once that is done, I can go back to applying the wifi patch you provided earlier.

elockman, 2023-03-14:
@Dharma-B did the buildOutput.txt log above help at all? Were you able to find any clues to the problem in it?

Can you share which Linux distribution you used to build from?

Dharma-B, 2023-03-14:
@elockman We are not using docker image to build.
We use the following build environment for a standard build
OS
```
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
```
Dependencies
```
sudo apt-get install gawk wget git-core git-lfs diffstat unzip texinfo gcc-multilib \
     build-essential chrpath socat cpio python3 python3-pip python3-pexpect \
     xz-utils debianutils iputils-ping python3-git python3-jinja2 libegl1-mesa libsdl1.2-dev \
     pylint3 xterm
```

Also, could you please try the solution given here ?
https://community.nxp.com/t5/i-MX-Processors/IMX8QXP-MEK-in-Yocto-5-15-71-rust-llvm-1-59-0-bb-do-compile/m-p/1599022

elockman, 2023-03-14:
Thank you @Dharma-B. I'll try the NXP solution for the rust-llvm error.

elockman, 2023-03-15:
@Dharma-B, thank you for the link. It looks like CircleCI is running out of RAM during the build. I'll be building the image locally later this week once a new desktop arrives.

elockman, 2023-03-22:
CircleCI was failing with a timeout on larger packages. I built kirkstone on a new desktop and added the patch, but I'm still not seeing either usb wifi with ifconfig. The devices are being reported with lsusb, as they had in the past.

```
root@sama5d2-xplained:~# lsusb
Bus 001 Device 003: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 004: ID 0bda:f179 Realtek Semiconductor Corp. 802.11n
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@sama5d2-xplained:~# ifconfig -a
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 38

can1: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 50

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.132  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::7284:f688:d1c8:3ddc  prefixlen 64  scopeid 0x20<link>
        ether fc:c2:3d:03:67:b0  txqueuelen 1000  (Ethernet)
        RX packets 203  bytes 28825 (28.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 34  bytes 5080 (4.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 28  base 0x8000

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 14  bytes 1902 (1.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14  bytes 1902 (1.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

sit0: flags=128<NOARP>  mtu 1480
        unspec 00-00-00-00-31-00-00-57-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
@Dharma-B what can I do next to troubleshoot this?

elockman, 2023-03-23:
@Dharma-B I got the wifi modules to work.

First, I added these image components:
```
IMAGE_INSTALL += "linux-firmware-ralink"
IMAGE_INSTALL += "linux-firmware-rtl8188"
IMAGE_INSTALL += "linux-firmware-mt7601u"
IMAGE_INSTALL += "linux-firmware-mt7601u-license"
# IMAGE_INSTALL += "kernel-module-mt7601u"
# IMAGE_INSTALL += "kernel-module-r8188eu"
```
Note, that the kernel modules are commented out.

Next, I created a wifi.cfg file and added it to the build with:
`SRC_URI:append:sama5d2 = " file://wifi.cfg"`
[wifi.cfg](https://github.com/linux4sam/meta-atmel/files/11055141/wifi.cfg.txt)

Thank you for your help, and sorry for the confusion with the CircleCI issues.
Closing this issue.

closed 2023-03-23 by elockman