linux4sam/meta-atmel #249: (Question) How to migrate to systemd

elockman, 2023-05-03:
I'm not having luck migrating from **System V** to **Systemd** in my yocto build.

1. Is there an online resource I missed on steps to do this?
2. Are there any good arguments not to migrate? (on an at91sama5d27 chip)

Dharma-B, 2024-01-18:
Hi @elockman,

We recently migrated to systemd from sysvinit, please check the latest commits in the kirkstone branch.

lukedukeus, 2025-05-05:
@Dharma-B  Thanks for the response.
Even with that set, it still uses systemv

mattwood-microchip, 2025-05-05:
Try this in your local.conf:

DISTRO_FEATURES_BACKFILL_CONSIDERED:append = " sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_dev_manager = "systemd"

lukedukeus, 2025-05-06:
@mattwood-microchip Unfortunatly it is still running systemv. do you have a minimal image demo that uses systemd?