linux4sam/meta-atmel #259: microchip-graphics-image errors

MarcinRobaczewski, 2023-09-02:
Hello.

As I compile "MACHINE=sama5d27-som1-ek-sd bitbake microchip-headless-image" it is a success.
When I compile "MACHINE=sama5d27-som1-ek-sd bitbake microchip-graphics-image" I get errors:

NOTE: arm-poky-linux-gnueabi-objdump -p /srv/marcin/yocto/linux4sam/kirkstone/build/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/libplanes/1.0.3+gitAUTOINC+d897de598d-r0/packages-split/libplanes/usr/lib/libplanes.so.1.0.1
ERROR: QA Issue: libplanes rdepends on libdrm-dev [dev-deps]
ERROR: Fatal QA errors were found, failing task.

NOTE: arm-poky-linux-gnueabi-objdump -p /srv/marcin/yocto/linux4sam/kirkstone/build/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/libegt/1.7-r0/packages-split/libegt/usr/lib/libegt.so
ERROR: QA Issue: libegt rdepends on cairo-dev [dev-deps]
NOTE: Checking Package: libegt-dev
NOTE: Checking Package: libegt-locale-de
NOTE: Checking Package: libegt-locale-fr
ERROR: Fatal QA errors were found, failing task.

It's a dependency problem for libplanes and libegt.
Any hints ? I am a beginner in Yocto.

gehariprasath, 2023-09-04:
Hello

Pls make sure you are checking out the right tags/commits in each layer as mentioned in the README file in meta-atmel. We will update the README file with clear instructions soon. For now, here's the information below

 1/ Clone yocto/poky git repository with the proper branch ready

git clone https://git.yoctoproject.org/poky && cd poky && \
git checkout -b kirkstone yocto-4.0.9 && cd -
 
 2/ Clone meta-openembedded git repository with the proper branch ready

git clone git://git.openembedded.org/meta-openembedded && \
cd meta-openembedded && git checkout -b kirkstone df452d && cd -
 
 3/ Clone meta-atmel layer with the proper branch ready

 git clone https://github.com/linux4sam/meta-atmel.git -b kirkstone
 
 4/ Clone meta-arm layer with the proper branch ready

git clone https://git.yoctoproject.org/meta-arm && cd meta-arm && \
git checkout -b kirkstone yocto-4.0.1 && cd -

MarcinRobaczewski, 2023-09-04:
Thank you, it worked.

gehariprasath, 2023-09-05:
Glad to know it worked. Closing this incident.

closed 2023-09-05 by gehariprasath