linux4sam/meta-atmel #261: cryptoauthlib_git.bb:do_install) failed with exit code '1'

elockman, 2023-09-12:
I am able to build the core image with:
`MACHINE=sama5d27-som1-ek-sd bitbake core-image-minimal`

But I run into a cryptoauthlib error when I run:
`MACHINE=sama5d27-som1-ek-sd bitbake microchip-headless-image`
I get the following error:
```
NOTE: Executing Tasks
ERROR: cryptoauthlib-1.0+gitAUTOINC+86b3046d2e-r0 do_install: ExecutionError('/home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/run.do_install.3808492', 1, None, None)
ERROR: Logfile of failure stored in: /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/log.do_install.3808492
```
I also see:
```
| cp: cannot stat '/home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/var/lib/cryptoauthlib/slot.conf.tmpl': No such file or directory
| WARNING: /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/run.do_install.3808492:148 exit 1 from 'cp -p /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/var/lib/cryptoauthlib/slot.conf.tmpl /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/var/lib/cryptoauthlib/0.conf'
| WARNING: Backtrace (BB generated script):
| 	#1: do_install, /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/run.do_install.3808492, line 148
| 	#2: main, /home/eric/Documents/repos/som1ek1/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/run.do_install.3808492, line 173
ERROR: Task (/home/eric/Documents/repos/som1ek1/poky/../meta-atmel/recipes-security/cryptoauthlib/cryptoauthlib_git.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5451 tasks of which 5444 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/eric/Documents/repos/som1ek1/poky/../meta-atmel/recipes-security/cryptoauthlib/cryptoauthlib_git.bb:do_install
Summary: There was 1 ERROR message, returning a non-zero exit code.
```
The image builds if I comment out the image recipe line 20:
`IMAGE_INSTALL:append:sama5d2 = " cryptoauthlib python3-cryptoauthlib p11-kit"`
or if I change it to:
`IMAGE_INSTALL:append:sama5d2 = " python3-cryptoauthlib p11-kit"`

Any thoughts @Dharma-B?
Is anyone else seeing this?

elockman, 2023-09-12:
Here is the log file from above.
[log.do_install.3808492](https://github.com/linux4sam/meta-atmel/files/12590109/log.do_install.3808492.txt)

Dharma-B, 2023-09-13:
Hi @elockman,
Does your system meet the minimum system requirements?

elockman, 2023-09-13:
@Dharma-B, yes I am building on a desktop dedicated to builds.
I needed to create the following bbappend file in order to get it to build. I'm not sure why, exactly. Does this point to a cmake issue?

```
do_install:prepend:sama5d2() {
    install -Dm 644 ${S}/app/pkcs11/slot.conf.tmpl ${D}${localstatedir}/lib/cryptoauthlib/slot.conf.tmpl
}

FILES:${PN} += "${sysconfdir}/cryptoauthlib "
```

Dharma-B, 2023-09-14:
Hi @elockman,
We didn't encounter such an issue, to reproduce the same could you please share the snapshot of bitbake build configurations?

mattwood-microchip, 2023-09-14:
@elockman,

I saw this a couple of days ago.  I had installed cryptoauthlib on the host PC while working with my meta layer which the only change I can think of that happened.  I manually removed the install, removed the tmp dir and rebooted the machine.  Then my build worked.

I think all of that is anecdotal as I don't see how having CAL installed on the host PC would cause any issues.

elockman, 2023-09-14:
I manually removed the build and started over, but it didn't resolve the
issue.  I didn't reboot. I can try that again.

elockman, 2023-09-14:
I am seeing the same issue. I even built in a new folder. These are the commands I run in the newly created folder:
```
git clone https://git.yoctoproject.org/poky && cd poky && git checkout -b kirkstone yocto-4.0.9 && cd -
git clone git://git.openembedded.org/meta-openembedded && cd meta-openembedded && git checkout -b kirkstone df452d && cd -
git clone https://github.com/linux4sam/meta-atmel.git -b kirkstone
git clone https://git.yoctoproject.org/meta-arm && cd meta-arm && git checkout -b kirkstone yocto-4.0.1 && cd -
cd poky
mkdir build-microchip
export TEMPLATECONF=${TEMPLATECONF:-../meta-atmel/conf}
source oe-init-build-env build-microchip
MACHINE=sama5d27-som1-ek-sd bitbake microchip-headless-image
```
This does not have any of my code. It should be a vanilla build.

I have a solution with the bbappend file shared above, so this is no longer blocking for me. Just looking to see if this helps others.